arduino/Arduino #4944: post_install.bat is not executed for a package_*_index.json file without signature

GrumpyOldPizza, 2016-05-11:
I am having this problem that "post_install.bat" is not executed, and the board manager throws the warning that it intentionally does so. 

How is one supposed to sign the package_*_index.json file ? Is there a public process for this ?

facchinm, 2016-05-11:
Hi @GrumpyOldPizza ,
this behaviour is intended. `post_install.bat` is only executed if it comes from a verified authority (right now, only "arduino" and "Intel" signed packages can trigger the post_install).
I'm sorry but there is no manual verification phase or "whitelist" (it would be too easy for an attacker to pass the verification when uploading the package for the first time and then change its behaviour)

closed 2016-05-11 by facchinm

GrumpyOldPizza, 2016-05-18:
Would it be possible to get a package file signed from arduino.cc ?