arduino/Arduino #4965: Problems with Desktop Integration on Linux (GNOME)

cookiengineer, 2016-05-18:
Hey guys,

I figured out a couple bugs when you download the current Arduino IDE from the website.
The install.sh script is buggy in two things:

- 1. The path has to be properly escaped with whitespaces ("Arduino IDE" as folder name will not put a "/blabla/Arduino\ IDE/arduino" in the resulting .desktop file
- 2. The command will fail because it's no binary. It's a shell script. That means "bash /path/to/arduino" whilst Terminal being false will only work. Never should've worked before.
- 3. The desktop file should've been placed inside ".local/share/applications" (as done already). With the above bugs fixed you can use it properly in GNOME 2 and 3.

How can I help on fixing these bugs? Would love to pick those up and to make a pull request.
Are pull requests even accepted here?

matthijskooijman, 2016-05-18:
1. I suspect that the script was only tested with a space-less path, then. Adding proper quoting makes senses to me (but it should then also take care of escaping any quotes in the folder name, of course).
2. A shell script with the +x permission set can also be executed like a binary. Are you sure that that bit isn't set? Perhaps when originally testing the generated desktop file, that bit was manually set (since I can't imagine this was never tested, IIRC the current .desktop generator was contributed through a PR).
3. So nothing to change here?

A pull request would be welcome. It might take a while to get it merged though - the Arduino devs are a bit short on time to process all contributions that come in, but something (presumably) simple and short like this can probably be merged quickly.

cookiengineer, 2016-05-18:
Regarding 2: The bash before the path had to be in there on my GNOME (3). It didn't work with the "arduino" shell script path and the script had +x chmod flag already.

per1234, 2017-07-06:
Possible duplicate of https://github.com/arduino/Arduino/issues/4581