arduino/Arduino #4969: WiFi101 Software won't compile

GregoryJH, 2016-05-19:
I just purchased a new WiFi101 board and put it on an Arduino Mega2560.  I downloaded the software WiFi101-Master.  It won't compile using the Arduino IDE and gives error messages.
E.g.
In member function uint8_t WiFiClass::begin(constr String&)

error:'const class String' has no member named 'c_str'

There are many more like this

facchinm, 2016-05-19:
Hi Gregory,
which version of the IDE are you running? It looks like your problem could be related with an ancient version of AVR core being used

GregoryJH, 2016-05-19:
I drive my Arduinos from a Raspberry Pi 2. The only Arduino IDE on this is 1.0.1.  I have several shields such as Grove and Adafruit Motor Board V2 and all the libraries for the sensors/controllers compile and work.  So I am puzzled why this is happening for the WiFi101 shield which is a genuine Arduino product.
Greg

facchinm, 2016-05-19:
IDE 1.0.1 is way too old and misses a lot of features introduced in the last 4 years; we just released a standalone version of the IDE for ARM systems, you can download it [here](https://www.arduino.cc/download_handler.php?f=/arduino-1.6.9-linuxarm.tar.xz) 
After the update your Wifi101 will work lika a charm :wink:

GregoryJH, 2016-05-19:
Does it work on a Raspberry Pi 2?
Greg

facchinm, 2016-05-19:
Yep!

GregoryJH, 2016-05-19:
Thanks.  I will see how it goes and let you know Greg

Chris--A, 2016-05-19:
@facchinm, what do you mean by standalone?

A new/stripped down IDE,
Or just not the typical linux build.

facchinm, 2016-05-19:
@Chris--A , it is a typical build out of Linux package manager (so with all the dependencies compiled in)

Chris--A, 2016-05-19:
Ah, cheers

GregoryJH, 2016-05-19:
What is the Linux command to extract the  Arduino-1.6.9-linuxarm.tar.xz?

facchinm, 2016-05-19:
`tar xvf` should do the trick

GregoryJH, 2016-05-19:
Thanks.  It unpacked and installed OK.  Now I just have to get it to run as typing arduino after $ runs the old IDE 1.0.1 version

GregoryJH, 2016-05-20:
I have sorted this and Arduino 1.6.9 now runs.  I just had to click on the executable Arduino file in the Arduino-1.6.9 file folder.  That's terrific because I thought I was stuck with Arduino 1.0.1 on an RPi.  Thanks  Greg

GregoryJH, 2016-05-20:
Just loaded the example ConnectwithWPA for WiFi101 and using Arduino 1.6.9 and it compiled and connected.  Thanks Greg

closed 2016-05-20 by cmaglie