arduino/Arduino #4974: Upload problem after upgrading Arduino AVR Boards to 1.6.10

hamid914, 2016-05-21:
I used Board Manager to upgrade Arduino AVR Boards  to version 1.6.10. After that, uploading any program to my MEGA board finished with no errors and it displays `Upload Done` but it is not functional on the board. Even Blink example does not work after upgrade.
I tried upgrade to 1.6.11 but the problem did not solved.
I tried downgrade to 1.6.9, after that all upload process finish with success and all programs are functional.
I reproduced this scenario on Windows and Mac with Arduino 1.6.9.

Why this problem happen?

facchinm, 2016-05-23:
Hi @hamid914 ,
could you paste here the sketch and the verbose output (also with compiler warnings set to "all")? Thanks

hamid914, 2016-06-19:
@facchinm Unfortunately I haven't access to the MEGA board for months I think. I will include more details later if the problem persists.

sandeepmistry, 2016-07-18:
I'm closing this for now, please re-open when you have more details.

closed 2016-07-18 by sandeepmistry