arduino/Arduino #4981: apparent typo in print.h:59

jsotola, 2016-05-23:
missing pointer name in line 59

line 59     size_t print(const String &);
.
.
.
line 71     size_t println(const String &s);

print.cpp has following ...

size_t Print::print(const String &s)
{
  return write(s.c_str(), s.length());
}

matthijskooijman, 2016-05-23:
In a function/method declaration, names for parameters are optional (they aren't actually used for anything, except perhaps as documentation). In a function/method definition (e.g. in the .cpp file) they are needed if the parameter is to be used.

So I guess this isn't a real bug, but just a small inconsistency that might or might not be worth fixing...

jsotola, 2016-05-24:
i see that now, if anything, the "s" should actually be taken out in line 71 to make it consistent with the rest of the header file

closed 2016-05-24 by jsotola