arduino/Arduino #4984: Copying text in the Serial Monitor window

johnsondavies, 2016-05-24:
If you select text in the Serial Monitor window and copy it to the clipboard, the last selected character gets omitted.

Tested on Arduino 1.6.8 on MacOS.

pierreverbakel, 2017-11-09:
I would realy (really!) like to have a Button below in the Serial Monitor window 'Copy to Clipboard'.  When having like 10.000 lines of continuous output it is nice look for specific moments of the run time, to see what my sketch did, measured, or decided. I really do not understand how this can be version 1.84 and not having such a Windows 1.0 function to be honest. Piece of cake to implement, loads of rewards.

oldmanegan, 2017-11-09:
On a Mac I do Cmd+A to select All then Cmd+C to copy to clipboard. There must be a windows analogue of this. Really need a button?

facchinm, 2017-11-09:
I just tested it on IDE 1.8.5 on OSX Sierra and it works correctly. Closing as solved, please reopen if the issue is still there on your configuration.

closed 2017-11-09 by facchinm