arduino/Arduino #4989: IDE colors not working properly (three Java coding glitches exist in v165 through v169)

Kenj-duino, 2016-05-26:
Three IDE color triggers do not work in v165 through v169: (have not worked since v164) 
These glitches may be hard to notice unless color theme is modified.  The functions still work fine. 
Copy the text below and paste it into the Arduino IDE to test/display these minor problems. 
 
1). Label: (color not responding to editor.label.style theme.txt or default.xml settings) 
Label_Color_Example: 
 
2). URL (underline and color not responding to editor.url.style theme.txt or default.xml) 
     Trigger/hyperlink still requires " // " proceeding the URL and underline is hard-coded in java. 
 www.arduino.cc/ // www.arduino.cc/  http://www.arduino.cc/  https://www.arduino.cc/ 
 
3). The 1-Star Block Comment  is tied to the wrong display color in theme.txt. 
/* MULTI-LINE COMMENT / */  // SHOULD USE THIS COMMENT COLOR 
//  ^-- these two colors should look alike --^ 
// The 1-star block comment should tie to editor.comment1.style 
/**  BTW, The 2-star Multi-line Block Comment Color is correctly tied to editor.comment2.style. */

lmihalkovic, 2016-05-26:
The syntax highlighting code in the new editor has issues. It's been there for some time and I pointed to other ways it manifests itself.

OldETC, 2018-02-09:
I am visually handicapped, and need this immediately.