arduino/Arduino #4992: IDE shadows warnings

NicoHood, 2016-05-28:
I enabled all warnings.
If I compile my program without errors, no warnings are displayed. If the code has got an error, those warnings appear on top of it.

This is twice bad:
* You will not notice any warnings until you get an error
* The error is hard to debug if you have tons of other warnings

Using latest IDE from git, linux.

facchinm, 2016-07-19:
Hi Nico, can you post an example sketch (and procedure) that triggers this behaviour? Please consider that compiling the same sketch twice silences the warnings (because the files emitting the warning are cached and not recompiled again). Or maybe the error you are introducing also creates the warnings :smile:

sandeepmistry, 2016-08-24:
Hi @NicoHood, any updates on this?

NicoHood, 2016-08-25:
I am not sure about this anymore. @facchinm is right about the caching. That might possibly the reason, and also the caching is good that it works like this. I will close it for now and reopen when I see this bug again + testcase.

closed 2016-08-25 by NicoHood