arduino/Arduino #4995: Inability to start IDE due to dialogue from #2719 for reasons unclear.

Ghostbird, 2016-05-29:
Today I updated my Arduino IDE to `ARDUINO 1.6.9 2016.05.10`

I tried to start it, but got the dialogue created to solve #2719: 
> Your copy of the IDE is installed in a subfolder of your sketchbook. Please move the IDE to another folder.

I was rather annoyed that it did not provide me with additional info that clarified why it was suddenly going wrong. I would have liked to see:
- The path to the IDE
- The sketchbook path

After as short search, I found that the sketchbook location was defined in .arduino/preferences.txt
I checked, and I concluded that my sketchbook folder was: `/home/gijsbert/sketchbook`.
My IDE is installed in `/home/gijsbert/Arduino`
I checked both paths and noticed that the sketchbook folder did not exist. It could be that it therefore defaulted to the part of the path that did exist, though that would be a bug in its own right.
I created the folder `/home/gijsbert/sketchbook`, and tried to restart Arduino IDE, but I still got the same error.
I ran:
```bash
ls -laR /home/gijsbert/Arduino | grep sketchbook
```
To verify that no links in the filesystem existed that would somehow link the IDE folder and the sketchbook folder together. (Doing this check the other way around wasn't necessary, because I just created the empty sketchbook folder.)

At this point I just decided that it was probably a report worthy bug. I'm sorry if I just overlooked something trivial, but if this happens to other people too, it should probably be looked into.

Ghostbird, 2016-05-29:
Even more silly. If the Arduino IDE is installed in `/home/username/Arduino` (which, as it turns out, is the default sketchbook location), you can start it `once`, since the fix for #2719 does not check whether the IDE location conflicts with the default value if no preferences.txt exists. However once you close it, and try to re-open it, the fix for #2719 kicks in, and you can no longer open the IDE.
It seemed logical to me to install the Arduino IDE into `/home/gijsbert/Arduino` and the sketchbook in `/home/gijsbert/sketchbook`. But apparently my logic triggered this whole issue.

It turns out the suggestion to move the IDE in the dialogue was indeed the logical solution when you thought about it from the point of view of the Arduino IDE writer.

per1234, 2020-08-01:
> this information is incorrect, and the preferences are located in the folder /home/gijsbert/.arduino15.

Thanks for pointing that out @Ghostbird! That documentation has since been updated.

Since the dialog mentioned here doesn't even work/exist anymore and we have a new issue where that situation as well as your excellent suggestion of making the dialog offer to change the sketchbook location is being discussed (https://github.com/arduino/Arduino/issues/10511), I'm going to go ahead and close this. I welcome your input over on https://github.com/arduino/Arduino/issues/10511 @Ghostbird.

closed 2020-08-01 by per1234