arduino/Arduino #4997: Dowloaded latest  IDE 1.6.9. as Linux 64 - Ubuntu - questions

VaclavSal, 2016-05-29:
My first venture into non Windows OS.

1. The IDE  "release" number is nowhere near of 1.6.9. 
     Is that normal?

2. There is no "Due" / ARM board option - hence the compiled / downloaded  program is bogus.
    No "ARM" subdirectory in files.

     Is that normal?
     Was support for Due dropped?

NicoHood, 2016-05-29:
Download the ide from arduino.cc. The packages of ubuntu are outdated.

VaclavSal, 2016-05-30:
I did - from arduino.cc "download" page. 
Can you be more specific what I did wrong? 
How am I suppose to know that it is not up to date?  
( I guess the version was a good indicator !)  
I suppose the "recompile / compile" documentation is also outdated. 
( It was pretty vauge year ago when I tryied to compile in Windows ) 
Can you confirm that?

VaclavSal, 2016-05-30:
Foudn this - is that current / updated ?

https://github.com/arduino/Arduino/wiki/Building-Arduino

NicoHood, 2016-05-30:
It is up to date. I was the one who updated it ;)
You can download their .tar.gz and the unextract it. Run sudo ./intall or just call the arduino binary from the command line `./arduino`.

VaclavSal, 2016-05-31:
OK, I managed to clone 
arduino-1.6.9-linux64.tar.xz

This is now in /Download directory 
If I do 
.\arduino I execute the wrong release 1.0.5 but it downloaded as 1.6.9. from Arduino.cc Download page. We agree that is bogus.

If I change to Download arduino-1.6.9-linux64.tar.xz and run sudo ./install arduino I get no such file.

I need to figure out which command works automatically - as you suggested ./arduino should work that way and which one would work from full command line. 
I'll work on that.

Thanks for you help, appreciate it. 
PS Do I need to EXTRACT all files from the download? 
You said  to "unextract" maybe I misunderstood.

NicoHood, 2016-05-31:
Yeah i mean extract. that is the reason why it does not work for sure. Also there is no backslash like this in linux. never use a backslash like this, it is wrong.

you can also `sudo apt-get purge arduino` to get rid of the old arduino installation which you really do not want to use. extract the package `tar -xf filename` and `cd folder` and then `./arduino` or `sudo ./install`

sandeepmistry, 2016-07-18:
I'm closing this due to lack of activity on @NicoHood's suggestion above.

> There is no "Due" / ARM board option - hence the compiled / downloaded program is bogus.
No "ARM" subdirectory in files.

The IDE no longer bundles support for the Due. You can install it via the board manager: `Tools -> Boards -> Board Manager`

closed 2016-07-18 by sandeepmistry