arduino/Arduino #5000: Arduino yun shield upload (ssh) and outputfile name.

jantje, 2016-05-30:
Hi
When uploading via the ssh uploader for the yun shield one first uploads the file (to a fix file name on the yun shield) and then run the remote upload command as specified in the platform.txt file.
For instance for the arduino sam (v1.6.8) the upload is
`tools.bossac_remote.upload.pattern=/usr/bin/run-bossac {upload.verbose} --port=ttyATH0 -U {upload.native_usb} -e -w -v -b /tmp/sketch.bin -R`

and for arduino avr (1.6.11) it is
`tools.avrdude_remote.upload.pattern=/usr/bin/run-avrdude /tmp/sketch.hex {upload.verbose} -p{build.mcu}`
The remote.upload pattern is clear but the copy command for ssh is not defined in the platform.txt and can only be know with text parsing or "hard coding"
for instance for avr the pseudo command is
`copy {build.path}/{build.project_name}.hex {sshtarget}://tmp/sketch.hex`

Where for sam it is
`copy {build.path}/{build.project_name}.bin {sshtarget}://tmp/sketch.bin`

The problem is that the platform.txt and the boards.txt do not specify the hex or bin part. To know whether to use hex or bin one needs to hard-code it or hard-code a parsing of the platform.txt.

Personally I think adding this information is far better than hard-coding.

There are plenty of ways this can be fixed. My personal preference goes to:
Creating a new tool for instance called avrdude_ssh (it can also be avrdude_remote)
switch to the correct tool based on remote upload or local upload which can be seen by the "com port" being local or remote
for remote avr ssh upload tools one has the following lines
```
tools.avrdude_ssh.upload.copy.source={build.path}/{build.project_name}.hex
tools.avrdude_ssh.upload.copy.target=/tmp/sketch.hex 
tools.avrdude_ssh.upload.pattern=/usr/bin/run-avrdude /tmp/sketch.hex {upload.verbose} -p{build.mcu}
```

for remote sam ssh uploads on has the following lines
```
tools.bossac_ssh.upload.copy.source={build.path}/{build.project_name}.bin
tools.bossac_ssh.upload.copy.target=/tmp/sketch.bin
tools.bossac_ssh.upload.pattern=/usr/bin/run-bossac {upload.verbose} --port=ttyATH0 -U {upload.native_usb} -e -w -v -b /tmp/sketch.bin -R
```

Can this be considered/discussed?

facchinm, 2016-05-31:
Hi @jantje ,
unfortunately the `copy.source` file is not the standard .hex but it's the `sketch+bootloader` version, which is not written as a file but lives in RAM and is copied to the YunShield using only Java code.
The relevant code is [here](https://github.com/arduino/Arduino/blob/8385aedc642d6cd76b50ac5167307121007e5045/arduino-core/src/cc/arduino/packages/uploaders/SSHUploader.java#L121) and it tries to handle all the corner cases (boards without bootloader, programmers without `_remote`) so it's quite difficult to make it generic and handle the differences in `platform` files.

jantje, 2016-05-31:
> it's quite difficult to make it generic and handle the differences in platform files.

Maybe it is but right now essential information is missing for supporting the ssh upload (so without adding the bootloader).
What I miss is:
1) what is the output file of the compilation?
2) What is the input file of the remote upload?

Basically I object to this piece of code https://github.com/arduino/Arduino/blob/8385aedc642d6cd76b50ac5167307121007e5045/arduino-core/src/cc/arduino/CompilerUtils.java#L45.
I ask to externalize the build output file to the platform.txt (or completely include it.
Other alternatives are:
`build.outputfile={build.path}/{build.project_name}.hex`
or
`build.outputfile.extension=hex`

closed 2018-05-25 by jantje