arduino/Arduino #5005: Building Linux - invalid Java release 8?

VaclavSal, 2016-06-02:
https://github.com/arduino/Arduino/wiki/Building-Arduino
I am following the above link and getting this show stopping / compile  error : 
compile:
    [javac] Compiling 138 source files to /home/a/Arduino/arduino-core/bin
    [javac] javac: invalid target release: 1.8
    [javac] Usage: javac <options> <source files>
    [javac] use -help for a list of possible options

So what is the correct JDK to use during cloning?

I use this command and tried to change to jdk 7 without success. 

**sudo apt-get install git make gcc ant openjdk-8-jdk**

There is a note on wiki to suggest / note errors. 

This wiki refers to Arduino IDE release, but calls it "JDK" .

I understand that things get misquoted / translated from native language to English , but ...
There must be a better way to  post instruction to "include path" -  the format used is misleading:  
cd /path/to/Arduino/build 

Since conventions are to include options in  [brackets] , why not 

cd [enter path to ] /Arduinoi/build  

Also emphasizing "use command " is little redundant since ALL data entered are commands.

NicoHood, 2016-06-02:
which os are you using? I think installing the java 8 should be fine. I dont remember those problems.

VaclavSal, 2016-06-02:
Ubuntu 14
How can I make sure I have Java 8  for Linux actually working?

sandeepmistry, 2016-07-18:
Hi @VaclavSal,

Have you tried running ` update-alternatives` to select the Java version More info: http://askubuntu.com/a/117443

facchinm, 2016-07-19:
Sandeep's answer is fine, closing as solved

closed 2016-07-19 by facchinm