arduino/Arduino #5006: Serial.AvailableForWrite missing in IDE 1.6.8 for MKR1000

Phil1942, 2016-06-02:
The Serial library for MKR1000 using IDE 1.6.8 seems to be missing the function AvailableForWrite. Confirmation or refutation requested.

matthijskooijman, 2016-06-03:
This sounds like a problem to report with the MKR1000 maintainers, I do not believe that that is part of the official Arduino IDE?

PaulStoffregen, 2016-06-03:
It is indeed part of Arduino's documented Serial API.  Every board should support it.

https://www.arduino.cc/en/Serial/AvailableForWrite

For boards without serial transmit buffering, a value of 0 or 1 should be returned, depending on the hardware bit that indicates if the port's buffer can accept the next byte.

sandeepmistry, 2016-06-03:
I agree with @PaulStoffregen's assessment, currently the SAMD core does not have serial transmit buffering, https://github.com/arduino/ArduinoCore-samd/issues/33 is open to track this, and I've submitted https://github.com/arduino/ArduinoCore-samd/pull/58 to resolve it.

I'll open a PR shortly for a simpler change to implement `Serial.availableForWrite` without the bigger buffer changes.

sandeepmistry, 2016-06-03:
I've submitted https://github.com/arduino/ArduinoCore-samd/pull/146 to cover the hardware serial on SAMD. However, this will only add it for `Serial1` on the MKR1000, as `Serial` uses CDC on native USB.

I've noticed the CDC implementation in both the SAM and SAMD core's are also missing `Serial_::availableForWrite`. The AVR implementation is a bit misleading to me, it returns `USB_SendSpace(CDC_TX)`, but `Serial_::write` uses `USB_Send` which will actually split up data into chucks based on `USB_SendSpace`. So, to me this means `availableForWrite` is actually unlimited (well bound by memory of course).

Any thoughts on this?

matthijskooijman, 2016-06-03:
> This sounds like a problem to report with the MKR1000 maintainers, I do not believe that that is part of the official Arduino IDE?

Heh, seems I totally missed that the MKR1000 board by Arduino. Scratch my comment :-)

PaulStoffregen, 2016-06-03:
> The AVR implementation is a bit misleading to me
> ....
> So, to me this means availableForWrite is actually unlimited (well bound by memory of course).

The AVR with on-chip USB hardware has a small dedicated memory for USB packet buffers.  Atmel calls it "DPRAM" in their datasheets.  It's completely separate from normal AVR memory and not addressable in any way other than special FIFO registers in the USB hardware.  The FIFO registers work based on other registers which configure the allocation of this special memory among all the USB endpoints.  Even within that small special memory, the entire space can't ever be allocated to any particular endpoint.  Different AVR USB chips support different numbers of endpoints and have varying rules for how the this memory can be allocated to them.

Any attempt to change availableForWrite() needs to very carefully consider these hardware limitations....

sandeepmistry, 2016-06-06:
After discussing this with @cmaglie, we've decided the SAM and SAMD `SerialUSB.availableForWrite()` needs to match the AVR behaviour of returning the number of bytes left in the current bank.

Since both implementation flush the bank on write, the return value will be the `EP size - 1`.

sandeepmistry, 2016-06-06:
I've submitted the PR's:

 * SAMD: https://github.com/arduino/ArduinoCore-samd/issues/148
 * SAM: https://github.com/arduino/Arduino/issues/5018

One note, the AVR version of `SerialUSB.availableForWrite()` currently returns 64 on the Leonardo, which is different from what I've implemented above. However, #4864 is open to change this to 63 (and it will also resolve #3946).

closed 2016-07-08 by agdl