arduino/Arduino #5008: Arduino 1.6.9 fails to build project

night-ghost, 2016-06-03:
while 1.6.6 works fine.

Steps to reproduce:
1.get sources for https://github.com/night-ghost/minimosd-extra
2.get sources for https://bitbucket.org/librepilot/librepilot/src
3.build LibrePilot's flight pars for any target - wee needs only UAVobjects
4. edit symlink MinimOsd_Extra/protocols/uavobjects to point to LibrePilot/code/librepilot/build/firmware/uavobjects

download MinimOsd_Extra.ino to IDE and try to compile. Error will be "openpilot.h not found". 

ps. Arduino.mk works fine too

facchinm, 2016-06-08:
Hi @night-ghost ,
could you provide a more simple way to test the failing sketch? A self-contained packet would be ok

night-ghost, 2016-06-08:
>failing sketch
Sketch is a code in one screen - but I say about rather big and complex project.

Yes I can do tar.gz from that 2 projects, but librepilot.tgz has near 2Gb size - do you want it?

Emmavico, 2016-06-20:
I have the same problem with the same projects in the projects book with arrays of arduino if you want this is the error
[error of arduino 1.6.9 .txt](https://github.com/arduino/Arduino/files/323146/error.of.arduino.1.6.9.txt)

facchinm, 2016-06-20:
Hi @Emmavico ,
your issue is totally different, yours are typos/code problems.
```
int buttons [0] =2;
``` 
should become 
```
buttons [0] =2;
```
```
Serial.Begin
```
should be corrected into 
```
Serial.begin
```
and 
```
Serial.printIn
```
is really 
```
Serial.println
```
(with the lowercase "L", not capital "i")

Next time please use the [forum](http://arduino.cc/forum) for this kind of problems, this issue tracker is meant to be used to collect bugs with the software itself, not programming help ;)

Emmavico, 2016-06-20:
ops i'm so sorry if I have disturbed you, thanks you ,bye

facchinm, 2016-06-20:
No problem at all :wink: