arduino/Arduino #5010: Arduino NANO Problems with uploading

AxelionEx, 2016-06-03:
This is the message that comes after i try uploading my code. i have an older mac but i was able to download the software. any suggestions would be awesome. This is my first robot.

Arduino: 1.6.9 (Mac OS X), Board: "Arduino Nano, ATmega328"

Sketch uses 1,066 bytes (3%) of program storage space. Maximum is 30,720 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2,039 bytes for local variables. Maximum is 2,048 bytes.
avrdude: ser_open(): can't open device "/dev/cu.adminsiMac-Bluetooth-PD": Resource busy
ioctl("TIOCMGET"): Inappropriate ioctl for device
Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

facchinm, 2016-06-06:
Hi @AxelionEx , did you select the correct port from the "port" menu? 
The `"/dev/cu.adminsiMac-Bluetooth-PD"` port in the error looks like the Bluetooth adapter embedded in any Mac. Does a port called "/dev/cu.usbmodem..." exist?

q2dg, 2016-06-07:
Well, it seems Arduino Nano isn't an official product...it doesn't appear on https://www.arduino.cc/en/Main/Products

AxelionEx, 2016-06-08:
No, i think it might be my mac. it's a mid 2007 model. i don't know why it wouldn't work it on it tho. i've looked at the differen't ports. Would downloading a package or library from arduino help solve it?

AxelionEx, 2016-06-08:
[image]
[image]

Ghostbird, 2016-06-12:
@AxelionEx Your Nano seems fine, that does not seem to be the issue. Did you look at this: https://www.arduino.cc/en/Guide/MacOSX#toc8 ?
It might help if you posted what's in that list.
Also, assuming I understand your set-up correctly, your diodes will not light up, because you did not connect the blue minus line on the breadboard to the Arduino GND.

agdl, 2016-07-25:
I'm closing this for lack of feedback. Reopen this if needed.

closed 2016-07-25 by agdl