arduino/Arduino #5012: Autoformat ignores tabsize preference

scottdarch, 2016-06-04:
I have `editor.tabs.size=4` in my preferences.txt which works fine if I hit <tab> but the autoformatter (cmd+T on OSX) ignores this setting and uses 2 spaces instead.

Verified this bug on both 1.6.8 and 1.6.9

JAG0991, 2016-06-10:
I'm looking within 1.6.10. The Auto Format tool uses a different configuration file called formatter.conf. It will look for a user defined formatter.conf within the same directory as your preferences.txt before using the default that can be found in the arduino/lib/ folder. The default formatter recommends that you copy it to the same directory as your preferences.txt and then edit it. Inside the formatter.conf is "indent=spaces=2". Changing this to 4 should hopefully make the Auto Format indent size the same as your editor.tabs.size.

jacob-jarick, 2016-09-23:
JAG0991 set me on the correct path. I use tabs always and 8 spaces (dont hate).

Exit arduino.
Create formatter.conf in your arduino conf directory (same location as preferences.txt
Add: indent=tab=8 (or your preferred size) then save.
edit preferences.txt

set the following 2 lines:

editor.tabs.expand=false
editor.tabs.size=8

done.

really should be an option in the GUI but its no hassle to edit a few config files.