arduino/Arduino #5017: SPI clockk and phase setting

shimizutko, 2016-06-06:
https://www.arduino.cc/en/Reference/SPI
does not say SPI clock(CPOL) & phase(CPHA) polarity ("1" or "0") to set MODE0-3.  
"1" means rising edge?
"0" means falling edge?

sandeepmistry, 2016-06-06:
Hi @shimizutko,

I believe it's a bit more complicated than this. Here's a from ["SPI Transfer Modes"](http://dlnware.com/theory/SPI-Transfer-Modes)

| Mode | Clock Polarity (CPOL) | Clock Phase (CPHA) | Data output on |
| ------- | ------------------------------| ----------------------------- | ------------------- |
| 0 | 0 | 0 | Falling edge |
| 1 | 0 | 1 | Rising edge |
| 2 | 1 | 0 | Falling edge |
| 3 | 1 | 1 | Rising edge |

The data capture will happen on the opposite edge of the the data output.

@SimonePDA any thoughts on how we could incorporate the above into the documentation page?

q2dg, 2016-06-07:
It is: https://www.arduino.cc/en/Reference/SPI

shimizutko, 2016-06-08:
sandeepmistry,

Thanks your comments.

I have confirmed the SPI signals by my oscilloscope.
Clock Polarity (CPOL)  is "level sensitive" as the Reference/SPI wrote.
Clock Phase (CPHA) is "edge sensitive"  as not on the Reference/SPI.

CPHA:0 means "Falling edge".
CPHA:1 means "Rising edge".

These are solutions.

Takao

agdl, 2016-07-11:
Done thank you!

closed 2016-07-11 by agdl