arduino/Arduino #5020: Board Manager Problem

kkunzle, 2016-06-07:
I just downloaded the new arduino IDE 1.6.9 and also just got an arduino 101 board.  I have tried to open the Board Manager, which it does, but gives an error saying that the download from the arduino website did not occur properly.  Therefore I can't get the board information I need to start using my new board.

Any Ideas?

facchinm, 2016-06-07:
Hi @kkunzle , could you paste here the error log?
BTW, we just pushed the updated Curie core (1.0.6), you could try downloading it and see if the problem disappears

kkunzle, 2016-06-07:
It says "Error downloading
http://downloads.arduino.cc/packages/package_index.json"

facchinm, 2016-06-07:
Can you reach that json file if you click on the link using a browser?

kkunzle, 2016-06-07:
Yes I can get to that link in the browser

facchinm, 2016-06-07:
Are you behind a proxy? In that case, try to set the same values in IDE's Preference menu

kkunzle, 2016-06-07:
I have tried changing the proxy settings but that doesn't change the result of the error.

kkunzle, 2016-06-07:
@facchinm The proxy has finally worked but now half way through the download of the Intel Curie Boards by Intel it gives a new error of
"Error Download https://downloadmirror.intel.com/25470/eng/arc-toolchain-windows-arcem-1.0.1.zip"

hmayorca, 2016-06-09:
Hi I have the same problem. My computer is running Windows 7 pro. I try in the new arduino IDE 1.6.9 and 1.6.8 version. I can´t update the boards library and i need to work with the Digital Sandbox for education.

facchinm, 2016-07-19:
Intel servers should be ok now, closing as solved

closed 2016-07-19 by facchinm