arduino/Arduino #5022: Cortex A8 binaries (for Artik) 32 bit compiler binaries are 64 bit

ShonkyCH, 2016-06-07:
http://downloads.arduino.cc/packages/package_arduino.cc_linux_index.json

refers to :

```
"host": "i686-linux-gnu",
              "url": "http://downloads.arduino.cc/libArduino/arm-cortex_a8-linux-gnueabihf-linux_i686-w.tar.gz",
              "archiveFileName": "arm-cortex_a8-linux-gnueabihf-linux_i686-w.tar.gz",
              "size": "50503386",
              "checksum": "SHA-256:398c82e4f41f89ee4d481d50db29b5e90457e1c771b40501ad3f03ed06685487
```

... for the tool chain, however the binaries in that tarball are all 64 bit, hence 32 bit linux dev machines can't compile. Apart from the build log and some of the libraries, the entire tar file is identical to the 64 bit one.

facchinm, 2016-06-08:
Hi @ShonkyCH ,
I'm working on recompiling the toolchain with the correct flags, but in the meantime I ask you to post only on the Artik forum, thanks :wink:

closed 2016-06-08 by facchinm

ShonkyCH, 2016-06-08:
Please explain? The "Online Help" from within the Arduino IDE links to https://github.com/arduino/libArduino/issues which doesn't exist but is under this github user/account.

There is no indication that this sort of issue should be posted in the Artik forum. Maybe the Online Help link should point to the Artik forums? Given that all the files come from arduino.cc how can I see the difference?

Also when this issue was posted, there has been little response apart from you asking about the binary. If the problem had been acknowledged a couple of days ago, I probably wouldn't have come here. The general slow response on the Artik forums is why I came here (along with the Online Help link above)

facchinm, 2016-06-08:
The "online help" points to a repo which is not yet open to public access. Please read again  ARTIK's Developer terms and conditions (which cover libArduino documentation) which specifies the "public beta" conditions (no links outside Artik forums etc).
The "slowness" on Artik forum can't be used as an excuse to bypass it in any way.
Thanks for understanding, the updated package will be online soon

ShonkyCH, 2016-06-08:
Thanks for fixing.  Not very helpful when an help link is inaccessible with no explanation. We can't read minds.

Please point me to the developer terms and conditions. I cannot find them apart from website T&C.