arduino/Arduino #5025: Building with arch

NicoHood, 2016-06-08:
@facchinm which dependencies do you need for arch? I've added some to the wiki, but i am not sure if those are correct:

https://github.com/arduino/Arduino/wiki/Building-Arduino#linux

I get the following error:
```
[alarm@alarmpi build]$ ant dist
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  Internal Error (os_linux_zero.cpp:254), pid=8240, tid=0x766de470
#  fatal error: caught unhandled signal 11
#
# JRE version:  (8.0_92-b14) (build )
# Java VM: OpenJDK Zero VM (25.92-b14 interpreted mode linux-arm )
# Core dump written. Default location: /home/alarm/hackallthethings/Arduino/build/core or core.8240
#
# An error report file with more information is saved as:
# /home/alarm/hackallthethings/Arduino/build/hs_err_pid8240.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#
/usr/bin/ant: line 6:  8240 Aborted                 (core dumped) /usr/share/apache-ant/bin/ant "$@"
```

I have to admit that I am currently using a pi (arm) and not x64. It might be an arm issue of ant and not the dependency instruction itself. Anything else I missed?

facchinm, 2016-06-09:
It looks like a Java on ARM problem (same [here](https://www.raspberrypi.org/forums/viewtopic.php?t=117968&p=802120)).
The solution posted [here](http://blog.timmattison.com/archives/2014/08/29/deal-with-os-linux-zero-dot-cpp-related-jvm-crashes-without-using-the-oracle-jvm/) should work.
We cross-compile the ARM version of the IDE is an OS where `ant` and java tools are stable enough :wink:

sandeepmistry, 2016-07-18:
Closing this for now, @facchinm's has linked to a work around above.

closed 2016-07-18 by sandeepmistry

NicoHood, 2016-07-18:
Correct. It works on arm now and I got a working PKGBUILD which i co maintain.