arduino/Arduino #5029: Problem updating Boards Managers

hmayorca, 2016-06-09:
**_**I have wIndows 7 64 and i receive the following error_**:**
I need to install Digital Sandbox board
Please Help

Error downloading http://downloads.arduino.cc/packages/package_index.json
java.lang.RuntimeException: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$19(ContributionManagerUI.java:151)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: Error downloading http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:113)
	at cc.arduino.contributions.packages.ContributionInstaller.download(ContributionInstaller.java:330)
	at cc.arduino.contributions.packages.ContributionInstaller.downloadIndexAndSignature(ContributionInstaller.java:305)
	at cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:283)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$19(ContributionManagerUI.java:147)
	... 1 more
Caused by: java.net.SocketTimeoutException: connect timed out
	at java.net.DualStackPlainSocketImpl.waitForConnect(Native Method)
	at java.net.DualStackPlainSocketImpl.socketConnect(DualStackPlainSocketImpl.java:85)
	at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:350)
	at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:206)
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:188)
	at java.net.PlainSocketImpl.connect(PlainSocketImpl.java:172)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at sun.net.NetworkClient.doConnect(NetworkClient.java:175)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:432)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:527)
	at sun.net.www.http.HttpClient.<init>(HttpClient.java:211)
	at sun.net.www.http.HttpClient.New(HttpClient.java:308)
	at sun.net.www.http.HttpClient.New(HttpClient.java:326)
	at sun.net.www.protocol.http.HttpURLConnection.getNewHttpClient(HttpURLConnection.java:1169)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1148)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:999)
	at sun.net.www.protocol.http.HttpURLConnection.connect(HttpURLConnection.java:933)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:170)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:111)
	... 5 more

facchinm, 2016-07-19:
Hi @hmayorca ,
sorry for the late response, do you still have this issue? The downloads servers are ok so you shouldn't encounter any problem in downloading the json (unless you are behind a proxy, which should work but you may need some tuning)

sandeepmistry, 2016-08-24:
I'm closing this due to lack of activity, please re-open with requested info. if you are still interested.

closed 2016-08-24 by sandeepmistry