arduino/Arduino #5032: Arduino Micro does not appear in /dev after OS X startup

notnot, 2016-06-12:
With the IDE 1.6.7 and 1.6.9 the Arduino Micros that i program do no longer appear in /dev after OS X startup. They only appear after unplugging and plugging them in again. This is very cumbersome and makes some applications impossible. 
With the older IDE 1.5.8 this used to work fine. Luckily I figured that out after lots of hair pulling! Has the
USB code that goes into the sketches changed? I don't know exactly when this bug has appeared, I can
only say somewhere between 1.5.8 and 1.6.7...

facchinm, 2016-06-13:
Well, between 1.5.8 and 1.6.7 the USB core was completely refactored to allow pluggable USB modules. So yes, the USB code into sketches changed, but this is the first regression we are aware of.
Are you doing some non-standard stuff with the USB in your sketch? Does it also disappear when you upload a blank sketch?

notnot, 2016-06-13:
No i am not doing non-standard things. The Arduino Micro is just communicating over /dev/usbmodem* , receiving commands and sending replies to the host computer (OS X), typically 333 times a second. I basically use it as a 1 - 12 channel ADC. 
At the moment the circuit with the Arduino is installed at a remote site, i can't test what happens when uploading a blank sketch. It is part of an art installation that will last a couple of months...

facchinm, 2016-06-14:
I understand... Could you share the sketch (also privately if you prefer) so we can test it with another micro + mac pc combo? 
Thanks

notnot, 2016-06-14:
Sure, here it is:

----------------------------------------------------------------------------------------------------------------------------------------
// input, jpad 2016
// n-channel analog -> digital input, output over USB

//// command/reply codes ///////////////////////////////////////////////////////

#define OK       uint8_t(0x00)
#define SAMPLE   uint8_t(0x01)
#define ERR      uint8_t(0xFF)

//// utility routines //////////////////////////////////////////////////////////

// read from serial port, blocks until data is available
inline uint8_t read() {
  while (Serial.available() == 0) {
    delayMicroseconds(100);
  }
  return Serial.read();
}

// write a 16-bit integer to the serial port, little-endian wire format
inline void write_uint16(uint16_t v) {
  Serial.write(uint8_t(v & 0xFF));
  Serial.write(uint8_t((v >> 8) & 0xFF));
}

//// commands //////////////////////////////////////////////////////////////////

bool DoCommand() {
  uint8_t cmd = Serial.read();
  switch (cmd) {
    case SAMPLE: {
        Serial.write(SAMPLE);
        DoSample();
        break;
      }
    default: {
        Serial.write(ERR);
        return false;
      }
  }
  return true;
}

void DoSample() {
  // get parameter: uint8 'number of channels'
  uint8_t nChannels = Serial.read();

  // validate parameter
  if (nChannels == 0 || nChannels > 12) {
    for (uint8_t c = 0; c < nChannels; c++) {
      write_uint16(0);
    }
    Serial.write(ERR);
    return;
  }

  // sample the requested number of channels
  for (uint8_t c = 0; c < nChannels; c++) {
    uint16_t v = analogRead(c);
    write_uint16(v);
  }

  // status code
  Serial.write(OK);
}

//// arduino framework functions ///////////////////////////////////////////////

void setup() {
  Serial.begin(230400);      // communicate at max speed
  analogReference(DEFAULT);    // AREF 5.0V
  //analogReference(INTERNAL); // AREF 2.56V

}

void loop() {
  if (Serial.available() > 0) {
    if (!DoCommand()) {
      // what to do here? stop responding?
    }
  }
}
----------------------------------------------------------------------------------------------------------------------------------------

PaulStoffregen, 2016-06-14:
Which version of OSX is this?

Many of the El Capitan versions appear to have bugs in the serial drivers.  :(

notnot, 2016-06-14:
Tested on OS X 10.0, and before i forget, on 10.6 this Arduino was never
reported in /dev

PaulStoffregen, 2016-06-14:
Of course Leonardo/Micro has never worked on Snow Leopard.  Apple's driver began supporting CDC IAD with Lion.

In fact, here's a message from their (then) driver developer regarding IAD descriptors from a Teensy-based test board I sent to Apple for testing.
http://lists.apple.com/archives/usb/2010/Nov/msg00040.html

notnot, 2016-06-14:
ah ok, i was unaware of that... i thought it could be related to same usb code issue

facchinm, 2016-06-15:
I tested the sketch with a Micro on both Mountain Lion and Yosemite and it works just ok...
@PaulStoffregen , the sketch doesn't use HID, so starting from PluggableUSB it "advertises" as IAD with only CDC capability... Could it be a cache problem (on OSX) since the board was previously saved as CDC+HID? On Win this problem is very frequent (and nasty)
@notnot , would you mind adding 
```
#include "Mouse.h"
#include "keyboard.h"
``` 
at the beginning of the sketch? This way the USB core becomes exactly as pre-1.6.5.

notnot, 2016-06-15:
@facchinm: that is odd. I tested it on 3 different macs, one of which had never seen any Arduino before. The 'keyboard setup" dialog didn't come up. 
Note that the issue i reported happend after a reboot only, only then the Micro isn't visible in /dev. Once it is unplugged and plugged again, it works, and stays visible also when you put the computer to sleep and wake it up.  

I have ordered a few new Micros, i'll do a test with including those headers soon and report back.

facchinm, 2016-06-15:
The fact that "keyboard setup" doesn't show up is a consequence of including HID capabilities only if needed.
I totally missed the "just after reboot" part. Could you test using IDE 1.6.9?

notnot, 2016-06-15:
@facchinm : IDE 1.6.9 suffers from the flaw