arduino/Arduino #5035: Library Not Updating

azdle, 2016-06-13:
I pushed some changes to my library [1]  before the weekend, but I'm still not seeing them pushed out to the IDE. Can anyone tell me what I did wrong that's causing my changes to not get pulled in? [2] is the tagged commit that bumps the library.properties version number.

[1] https://github.com/exosite-garage/arduino_exosite_library
[2] https://github.com/exosite-garage/arduino_exosite_library/commit/30b9151d829aec22ca5f4b2ee7ba8a02524669c8

facchinm, 2016-06-13:
Hi Patrick,
the Jenkins job which handles the libraries update had some vacation this weekend :smile:  I've just restarted it so your library should be updated in a matter of an hour

azdle, 2016-06-13:
Ah, good to know. Thanks. I'll close the issue when I see it updated.

azdle, 2016-06-13:
All good now. 👍

closed 2016-06-13 by azdle