arduino/Arduino #5039: New library: FirmataWithDeviceFeature

finson-release, 2016-06-14:
Repo: [https://github.com/finson-release/FirmataWithDeviceFeature](https://github.com/finson-release/FirmataWithDeviceFeature)

FirmataWithDeviceFeature is a clone of [ConfigurableFirmata](https://github.com/firmata/ConfigurableFirmata), with the addition of one more feature: DeviceFirmata.

This feature provides a Firmata interface to the device drivers on an Arduino.  [Pull request 52](https://github.com/firmata/ConfigurableFirmata/pull/52) has been submitted for ConfigurableFirmata, but until it is accepted this repo will serve.

See also:

[Luni - Arduino device driver framework.](https://github.com/finson-release/Luni](https://github.com/finson-release/Luni)
)

[LuniJS - Javascript client](https://github.com/finson-release/LuniJS)

Wiki: [https://github.com/finson-release/finson-release.github.io/wiki](https://github.com/finson-release/finson-release.github.io/wiki)

agdl, 2016-06-14:
Hi please review tags with what is written in the library.properties

finson-release, 2016-06-14:
Arturo:

Done, thank you.  I created a new tag, v2.9.0, in the repo.  I think this  
resolves the issue; if not, please let me know.

Doug