arduino/Arduino #5040: Library Manager out of date?

connornishijima, 2016-06-14:
I've been having some issues getting the Library Manager to stay current with the releases of my Volume library. [https://github.com/connornishijima/arduino-volume](https://github.com/connornishijima/arduino-volume)

[http://downloads.arduino.cc/libraries/library_index.json](http://downloads.arduino.cc/libraries/library_index.json) (CTRL+F search "nishijima")

The versions 1.0.0, 1.0.2, and 1.1.0 have been automatically picked up by your system, but these versions aren't options in the Library Manager, I can only download 1.0.0. (Latest version is 1.1.1, but the cron job for fetching new versions for the library index hasn't run yet for that one.)

Am I missing something? :(

facchinm, 2016-06-14:
Hi @connornishijima ,
the library manager seems to work correctly in my case (I didn't have the library installed and it show all the versions)
[image]
Can you try to move your library out of the sketchbook to see if the choices appear?

connornishijima, 2016-06-14:
Whoops. I was using Arduino 1.6.7 instead of 1.6.9. My bad! Thanks anyways.

closed 2016-06-14 by facchinm