arduino/Arduino #5042: Arduino IDE selects wrong board package toolchain

vshymanskyy, 2016-06-15:
The Arduino IDE (or arduino-builder), selects the wrong toolchain when some additional platforms are installed through Boards Manager.

For example, if I install both Digistump Oak and ESP8266 community packages, it selects
```.arduino15/packages/digistump/tools/xtensa-lx106-elf-gcc``` instead of
```.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc``` when compiling for regular ESP8266.

It causes more problems like this:
I'm trying to build "Blink" example for MKR100 (after installing RedBear Duo support package):
It fails, because IDE selects ```~/.arduino15/packages/RedBear/tools/arm-none-eabi-gcc/4.9-2015-q3``` toolchain, and it's not compiling correctly.

Same for some other boards combinations.

vshymanskyy, 2016-06-15:
https://github.com/arduino/Arduino/issues/4593 - might be relevant.

cmaglie, 2016-06-15:
May you post a link to the boards managers packages and the IDE version you're using?

vshymanskyy, 2016-06-15:
How to reproduce:

1. Install Arduino IDE v1.6.9
2. Install MKR1000 board support from the Boards Manager
3. Check that "Basics->Blink" example compiles fine for MKR1000 board.
4. Configure additional Boards Manager source: https://redbearlab.github.io/arduino/package_redbear_index.json
5. Install RedBead Duo package the Boards Manager
6. "Basics->Blink" example compilation **fails for target MKR1000 board**.

facchinm, 2016-06-15:
This is probably solved by https://github.com/arduino/Arduino/pull/4795, would you mind giving the @ArduinoBot build a try?

PR #5199 (2016-08-03): Fix tools discovery by strictly following json data when available
Should fix #5042 and #5168.

(in particular #5168 would need more work to implement my proposal of `coreDependencies` field into package_index.json to be a complete solution, IMHO)

See also #5197.

/cc @facchinm @PaulStoffregen @matthijskooijman @per1234 @jantje
closed 2016-08-03 by cmaglie