arduino/Arduino #5043: Allow importing a ZIP library package with multiple libraries inside

vshymanskyy, 2016-06-15:
I'm core developer of the Blynk library.
It supports lot's of hardware types and boards (more than 60!), but it also means it has a bunch of dependency libraries.
We run into a situation, where it would be much easier  to import our library, bundled with some dependencies in a single ZIP file.

For example, to be able to import all libraries from this ZIP:
https://github.com/blynkkk/blynk-library/releases/download/v0.3.7/Blynk_v0.3.7.zip

We could think about adding a special marker file in the root of the package, so IDE can reliably detect a multi-library package.

vshymanskyy, 2016-06-21:
ping....

bperrybap, 2016-06-30:
Rather than inventing something new for this, perhaps it is time to look at extending the library manager to support platformio which has a .jason file that can specify dependencies.
It would also be a win for the Arduino support team as they could start using that to add new libraries rather than having to manually weed through Arduino issues.

joshualyon, 2017-05-14:
This would also be helpful for trainings, workshops, and labs where it might be necessary to import several different libraries.