arduino/Arduino #5044: Behaviour of Sketch -> Include Library

vshymanskyy, 2016-06-15:
I'm core developer of the Blynk library.
It supports lot's of hardware types and boards (more than 60!), and we have a separate .h file for some board types.
So, after user selects ```Sketch -> Include Library -> Blynk```, it gets all these files included!

```
#include <WidgetTerminal.h>
#include <WidgetLED.h>
#include <BlynkSimpleEsp8266_SSL.h>
#include <BlynkSimpleSerialBLE.h>
#include <BlynkSimpleParticle.h>
#include <WidgetLCD.h>
#include <BlynkSimpleTI_TivaC_Connected.h>
#include <BlynkSimpleCC3000.h>
#include <BlynkWidgets.h>
#include <BlynkSimpleSimbleeBLE.h>
#include <BlynkSimpleWildFire.h>
#include <BlynkSimpleTI_CC3200_LaunchXL.h>
#include <BlynkSimpleUIPEthernet.h>
#include <BlynkSimpleEthernetV2_0.h>
#include <WidgetBridge.h>
#include <BlynkSimpleYun.h>
#include <BlynkSimpleWifi.h>
#include <BlynkSimpleEthernet2.h>
#include <BlynkApiParticle.h>
#include <BlynkSimpleEsp8266.h>
#include <WidgetSD.h>
#include <BlynkSimpleRFduinoBLE.h>
#include <BlynkApiArduino.h>
#include <BlynkSimpleShieldEsp8266.h>
#include <BlynkSimpleUserDefined.h>
#include <BlynkSimpleMKR1000.h>
#include <BlynkSimpleWiFiShield101_SSL.h>
#include <BlynkSimpleWiFly.h>
#include <BlynkSimpleEthernet.h>
#include <BlynkSimpleRBL_CC3200.h>
#include <WidgetRTC.h>
#include <BlynkSimpleRedBear_Duo_BLE.h>
#include <BlynkSimpleRedBearLab_BLE_Nano.h>
#include <BlynkSimpleStream.h>
#include <BlynkSimpleIntelEdisonWiFi.h>
#include <BlynkSimpleCurieBLE.h>
#include <BlynkSimpleEnergiaWiFi.h>
#include <BlynkSimpleRBL_WiFi_Mini.h>
#include <BlynkSimpleTinyDuino.h>
#include <BlynkSimpleLinkItONE.h>
#include <BlynkSimpleWiFiShield101.h>
#include <BlynkSimpleSerial.h>
#include <BlynkParticle.h>
#include <BlynkSimpleRedBear_Duo.h>
```

This is very wrong in our case. Is there any way to suppress this behaviour?
Maybe, indicate which file needs to be included in the ```library.properties```, or something similar?

cmaglie, 2016-06-15:
> Maybe, indicate which file needs to be included in the library.properties, or something similar?

It could be, another possibility is to include just the file named as the library, for example `Blynk.h`, but I can't see this file in the list above... which are the files that are supposed to be included in your case?

vshymanskyy, 2016-06-15:
If I create Blynk.h - it also imports all of the header files.
Our library provides many examples for each hardware type so this means - different headers may be needed. But including all of the headers is frustrating for users.

If a single Blynk.h was induced - it could just generate an compile-time error, pointing our users in right direction. Maybe there is some other option...

Anyway - including all headers (in random order?) won't work in most cases.

cmaglie, 2016-06-15:
> Anyway - including all headers (in random order?) won't work in most cases

the ordering should not be relevant, and this method worked for most of the currently exisisting libraries.

BTW, I'm asking about a possible fix: 

1) including just, say, `Blynk.h` would work for you?

2) if the answer to 1 is no, what could be a possible fix that works for you?

vshymanskyy, 2016-06-15:
Yes, it would be much better. We can add a special flag to library.properties to enable this, not to influence other libraries.

vshymanskyy, 2016-06-15:
Yes, it would be much better. We can add a special flag to ```library.properties``` to enable this, not to influence other libraries.
Something like ```include=Blynk.h```

closed 2016-07-01 by cmaglie

cmaglie, 2016-07-01:
@vshymanskyy the library.properties now supports the `includes` field, with the behaviour you mentioned. It will be released with IDE 1.6.10 and from the next hourly build.

PR #5066 (2016-07-01): "Include library" now checks for `includes` property
A new property "includes" has been added to library.properties.
This property contains a comma-separated list of the files to be included when
the user selects the "Include library" command on the Arduino IDE.

If the property is missing the old behaviour is used.

Fixes #5044 

/cc @vshymanskyy @matteosuppo