arduino/Arduino #5048: Library register request: Thing+

yhjo8478, 2016-06-17:
Hello. I made Arduino library.
This library helps connecting your Arduino board to Thing+ cloud easly and fast.
Please add my library to Library manager.

This is github url : https://github.com/daliworks/arduino_library

Thanks advance.

agdl, 2016-06-17:
Hi, I'm sorry but you library can't be added since the release is not compliant with standard arduino library

https://github.com/daliworks/arduino_library/tree/v1.0.0

Please fix it and consider to rename the repository in thingplus instead of arduino_library

yhjo8478, 2016-06-17:
Thank you for fast reply.
I changed library name from arduino_library to Thingplus and i re-release it.
Please check my changes from https://github.com/daliworks/arduino_library

agdl, 2016-06-17:
Done thank you! (Please delete unused release)

closed 2016-06-17 by agdl