arduino/Arduino #5051: Arduino Serial Light firmware

CRImier, 2016-06-18:
I've got an Arduino Serial Light USB-UART converter which blew up due to 12V applied. I had replaced the IC since I had a spare, and the 16MHz crystal as well. However, I can't find the right firmware for it. What'd be the firmware to use? Can I get a firmware image that'd make this dongle enumerate as the Serial Light adapter (that is, the original firmware)?

agdl, 2016-06-22:
@CRImier you can find it here
https://github.com/arduino/Arduino/tree/master/hardware/arduino/avr/firmwares/atmegaxxu2/arduino-usbserial

NicoHood, 2016-06-22:
Or you could also use it with hoodloader2 and add some enhanced functionality:
https://github.com/NicoHood/HoodLoader2

CRImier, 2016-07-01:
Oh, sorry, I fixed my adapter and forgot about this. Regardless, the question still remains - I flashed one of the firmwares that you linked and it seems to work. However, it enumerates as "Arduino Uno", which is weird and confusing. Is there a USB Serial Light firmware, specifically?

agdl, 2016-07-01:
@CRImier this is a discontinued product. If you want to flash it with the right VID/PID you should compile it again changing the VID/PID with the USB2Serial one

CRImier, 2016-07-01:
@agdl Sad to hear that "discontinued" means "firmware files not available". I'll try that, found PID in the Windows driver. Thank you!

closed 2016-07-01 by CRImier