arduino/Arduino #5053: arduino ide(i tried all version ) doesn't compile arduino due( even blink project ) .

MehmetKALA, 2016-06-19:
i have all arduino ide versions and all sam boards version but it doesn't comple even a simple project. No error shown, just type 'Compling sketch' and nothing. i tried reinstalling arduino ide and formatting pc many times. OS is Windows 10.( later formats, one time it compiles but than doesn't compile)

facchinm, 2016-06-20:
Hi @MehmetKALA ,
could you activate "verbose compilation" from Preferences menu and paste here the output? Thanks!

Chris--A, 2016-06-20:
You could also try running the IDE in compatibility mode for windows 8 or 7

PaulStoffregen, 2016-06-20:
Maybe you're seeing a "no error shown" similar to this video?  (compile fails at 46:05)

https://youtu.be/vopvXHW0Bok?t=46m5s

MehmetKALA, 2016-06-21:
thank you all, when windows 8  compatibility mode, it works, today i realized that it was compiling but it takes too long such as 5 minute for blink example. now no problem , everything is okey, thanks for that :).

closed 2016-06-21 by MehmetKALA

facchinm, 2016-06-21:
If it takes such a long time to compile it's probably an antivirus issue, as it tries to scan every temporary file produced by the build. 
Disabling it temporarily should tell you if I'm right, and he's the culprit you can add more fine-grained whitelist rules for `arduino-builder` and the temporary folders