arduino/Arduino #5054: The firmata should have a China name ?  阿朵诺?

nisen, 2016-06-19:
The arduino to now there is no Chinese name, So many people do not know arduino.
I introduce him difficult, if android so easy.   
arduino is so good for mcu.
In the future China, MCU chip should do with knitting socks almost universal :).
I want introduce the protocol to China maker community and MCU Factory.
We named it 阿朵诺

The reason link is <https://github.com/osokay/Arduino/blob/master/%E4%B8%AD%E6%96%87%E5%90%8D.md> in Chinese 

## Reference

 * firmata  冷杉脉络 <https://github.com/firmata/protocol/issues/63>
 * git  吉塔 <https://github.com/osokay/gitDialect>
 * android 安卓

soundanalogous, 2016-06-22:
@nisen please keep this discussion within the Firmata repository since Firmata is not hosted by Arduino nor does the Arduino team work on it.

agdl, 2016-06-23:
I'm closing this. @nisen follow the @soundanalogous suggestion

closed 2016-06-23 by agdl