arduino/Arduino #5058: Serious issue with Library manager allows forked libraries to supercede original library

soundanalogous, 2016-06-21:
I noticed today that the [ConfigurableFirmata library](https://github.com/firmata/ConfigurableFirmata) I maintain is updatable via the library manager to v2.9.0. Strange thing is I've never published a version 2.9.0. Latest version is [v2.8.2](https://github.com/firmata/ConfigurableFirmata/releases). Looking into this, it appears that [a fork of ConfigurableFirmata was renamed and then published separately](https://github.com/finson/FirmataWithDeviceFeature/releases). In the library manager, this ends up superseding the original library. I'm not even sure how this is possible, but I seems to be a serious bug and should be prevented going forward.

finson-release, 2016-06-21:
As the creator of the forked repo, I'd sure like to know ASAP how to unwind this.  Jeff tells me that anyone attempting to update ConfigurableFirmata now gets a copy of my fork, not the mainline version.  That is NOT the intention.

cmaglie, 2016-06-21:
I confirm that there is surely a bug on the indexer, I'm working on it.
Until I understand what's happening, I'll temporary remove the forked library by @finson-release to not cause further confusion.

finson, 2016-06-21:
FYI.

I requested that my fork be included in the library manager with issue #5039.  At that time the latest tag was 0.9.0 and the library had the new name (FirmataWithDeviceFeature) in library.properties.  However, the repo contains other, earlier tags that were created when library.properties in my fork still reflected the original name (ConfigurableFirmata) and the original version (2.8.2).

I changed the name and version before I requested the addition to the library manager, but I did not delete the older tags.  Perhaps the indexer tied my fork to ConfigurableFirmata based on those early tags, and then did not detach it when the name changed?

In any event, I can delete all the pre-library-manager-request tags if that would resolve this (or at least be a workaround).  No action until directed, though!

soundanalogous, 2016-06-21:
I would use caution deleting tags (especially pre 2.9.0 tags) now in case that would result in those versions in the ConfigurableFirmata library disappearing from the Library Manager.

finson, 2016-06-22:
!!  
As I say, I will not take any action until explicitly directed, for just this sort of reason.

agdl, 2016-06-22:
@soundanalogous I'm tagging you here since I can't in the other issue. Please have a look to this
https://github.com/arduino/Arduino/issues/5054

cmaglie, 2016-06-22:
I've already (temporary) removed the library made by @finson from the library manager so at the moment there are no "ghost" releases of ConfigurableFirmata. As I said the bug is in our indexer not in @finson's library, but it turns out to be quite tricky bug and needed a bit of refactoring.

I'm almost ready to deploy the fix, I just need a bit more time :-)

finson-release, 2016-06-22:
Cristian:

No hurry at my end.  I'm the only user of my fork at the moment, and it's  
available from the repository if others want to try it.  Take the time you  
need.

Thanks.

Doug

cmaglie, 2016-06-22:
Ok, I've fixed the indexer. This update has removed a lot of duplicates and revealed a pair of "hidden" releases (where the original library "covered" the fork, basically the opposite of what happened to ConfigurableFirmata).

Previously it was possible to change name between releases, **now this is no more allowed** and libraries must keep the name they used when registered. If the name in `library.properties` is changed the indexer will override this setting and always use the name used during registration.

There is a bunch of libraries that have already changed their name for some reason, here the list:

| Repository | Names (**official name in bold**) |
|------------|----------------------|
| sparkfun/SparkFun_Micro_OLED_Arduino_Library | Micro OLED Breakout, **SparkFun Micro OLED Breakout** |
| sparkfun/SparkFun_MicroView_Arduino_Library | MicroView, **SparkFun MicroView** |
| sparkfun/SparkFun_SX1509_Arduino_Library | **SX1509 IO Expander**, SparkFun SX1509 IO Expander |
| sparkfun/SparkFun_LSM9DS0_Arduino_Library | LSM9DS0 Breakout, **SparkFun LSM9DS0 Breakout** |
| sparkfun/SparkFun_ISL29125_Breakout_Arduino_Library | **SparkFun ISL29125 Breakout**, ISL29125 Breakout |
| sparkfun/SparkFun_AD5330_Breakout_Arduino_Library | **SparkFun AD5330**, AD5330 Breakout, Sparkfun_AD5330 |
| sparkfun/SparkFun_TSL2561_Arduino_Library | SparkFun_TSL2561_Luminosity_Sensor_BOB_Arduino_Library, **SparkFun TSL2561** |
| sparkfun/SparkFun_Line_Follower_Array_Arduino_Library | RedBot Line Follower Bar Arduino Library, **SparkFun Line Follower Array** |
| marcoschwartz/LiquidCrystal_I2C | LiquidCrystal_I2C, **LiquidCrystal I2C** |
| panStamp/thermistor | thermistor, **THERMISTOR** |
| panStamp/mma8652 | SWAP, **MMA8652** |
| panStamp/sram | SWAP, **SRAM** |
| willie68/RCReceiver | MCStools, **RCReceiver** |
| x2bus/EnergyBoard | MAX78630, **EnergyBoard** |
| stevemarple/SoftWire | AsyncDelay, **SoftWire** |
| somsinchai/IBot | WebServer, **IBot** |
| feilipu/Arduino_RTC_Library | AVR Standard C (C90) Time Library, **AVR Standard C Time Library** |
| winlinvip/SimpleDHT | Simple DHT sensor library, **SimpleDHT** |
| orgua/iLib | iLib, I2C-Sensor-Lib (iLib), **I2C-Sensor-Lib iLib** |
| orgua/OneWireHub | OneWire-Slave-Hub, **OneWireHub** |
| gmag11/NtpClient | NtpClient, **NtpClientLib** |
| paulo-raca/YetAnotherArduinoPcIntLibrary | **Yet Another Arduino PcInt Library**, Sodaq_PcInt |
| tomstewart89/BasicLinearAlgebra | Matrix, **BasicLinearAlgebra** |

for these libraries I've selected the name used in the **latest release**, in bold in the table above, as the official name. Even for those, starting from now, is no more possible to change name without admin intervention.

Here the resulting changes in the library_index.json: https://gist.github.com/cmaglie/987c0d73b469fd0707c1618abc7fafa1

@finson 
now your library is correctly listed, you can remove the old releases in your fork if you think it's appropriate.

closed 2016-06-22 by cmaglie

finson, 2016-06-22:
Thanks Cristian.  Will do.  Doug

soundanalogous, 2016-06-22:
Thanks Christian for your timely attention and solution to this issue!

matthijskooijman, 2016-06-22:
> If the name in library.properties is changed the indexer will override this setting and always use the name used during registration.

I wonder if this is the right approach - perhaps a release with a different name should b ignored by the indexer? Imagine a fork that synchronises with the original, including tags, which would now be published with a combination of the fork's name and the original's version.

Actually ignoring these would mostly work if there is some kind of feedback mechanism from the indexer to the library authors, but perhaps this might be a good idea anyway (e.g. a webpage that lists, per repository/library, an indexer log or status display? That might also help to let people self-diagnose other library problems.