arduino/Arduino #5059: Arduino IDE 1.6.9 64-bit for Linux -- Compiling Errors due to hard-coded paths pointing to /home/jenkins/jenkins/...

HP-Sparks, 2016-06-21:
Hi,
I just installed the Arduino IDE v1.6.9 64-bit on a Linux Mint 17.3 Cinnamon 64-bit machine.

Trying to compile an old project involving .h files for which I have to search for hints on many different Web sites since many different versions exist (...), I'm also facing the following error messages:

`Arduino: 1.6.9 (Linux), Board: "Arduino/Genuino Uno"

*errors.errorString invalid syntax
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/i18n/errors.go:19 (0x4b0d72)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/arduino-builder/main.go:362 (0x403ba0)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/arduino-builder/main.go:268 (0x402a3a)
/opt/go/src/runtime/proc.go:63 (0x4151d3)
/opt/go/src/runtime/asm_amd64.s:2232 (0x43b271)

Error compiling for board Arduino/Genuino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

Seems like hard-coded paths references as there is No "Jenkins" user on this system. (see "/home/jenkins/..." in above error messages)

Trying to compile a basic example like "blink" doesn't show any error.

Any idea / suggestion for a proper workaround until this bug get fixed ?

Thanks in advance.

Best regards,
  HP_

HP-Sparks, 2016-06-21:
The references to "/home/jenkins/... ", etc... disappeared once I removed a space in the name of an intermediary folder.
So, this may me a useful hint for a bug fix...

Best regards,
  HP_

matthijskooijman, 2016-06-21:
The "jenkins" stuff is just a reference to the directory where the arduino-builder utility is compiled, it is just informative. The actual error is `*errors.errorString invalid syntax`, though I'm not sure what that actually means. @cmaglie, saw this one before?

> The references to "/home/jenkins/... ", etc... disappeared once I removed a space in the name of an intermediary folder. (didn't spotted it on this 10" NetBook machine...)

Does that actually fix compilation? Or does it only change the error message? If the latter, what error message remains? Also, as suggested before, make sure to enable verbose compilation for more details.

cmaglie, 2016-06-21:
> @cmaglie, saw this one before?

never saw that

I guess it's worth looking how the builder handles errors here:

```
*errors.errorString invalid syntax
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/builder/i18n/errors.go:19 (0x4b0d72)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/arduino-builder/main.go:362 (0x403ba0)
/home/jenkins/jenkins/jobs/arduino-builder-linux64/workspace/src/arduino.cc/arduino-builder/main.go:268 (0x402a3a)
/opt/go/src/runtime/proc.go:63 (0x4151d3)
/opt/go/src/runtime/asm_amd64.s:2232 (0x43b271)
```

maybe the file path is passed internally as a format string?
BTW I think this is an arduino-builder issue, probably it's worth moving to the arduino-builder repo.

HP-Sparks, 2016-06-21:
@matthijskooijman, No more problem after removing that space.
Using the verbose optoin for compilation shows me what command line the IDE is using. Everytime, now, I can me my own username instead of "jenkins".

So, may be it is a quick and dirty fix to be addressed by this developer(?) to make Arduino IDE for Linux (only?) able to accept spaces in file/folder/path names. But, are spaces well accepted by other components like avrdude, avrgcc, etc... ?
Any way, this is out of my user's scope, for now... :-P ;-)

matthijskooijman, 2016-06-21:
So the actual problem is that a space in a filename or directory somewhere breaks compilation. Where exactly did you have this space?

HP-Sparks, 2016-06-21:
@matthijskooijman , At first, I just open the .ino file of the "TFT_Oscillo" project from a folder named "new folder" in my ~/Downloads/ directory.
Then, I thought that I should bring it into my "sketchbook" folder... From there, no more "jenkins" error.
Next, I wanted to check something I faced many years back (2009 or 2010) about spaces in file/folder/path names... So, I gave a try by renaming "TFT_Oscillo" folder into "TFT Oscillo".
Then, opening the .ino file again gave me a warning saying "must create a folder 'TFT_Oscillo' and move files inside" from the IDE. I chose to ignore it and keep doing with the folder name containing the space.
Then, checking/compiling gave me the "jenkins" errors... => spotted a good trigger to reproduce the problem. ;-)

HP-Sparks, 2016-06-22:
@matthijskooijman , @cmaglie , Did you manage to reproduce the problem or is it only on my system/PC ?

matthijskooijman, 2016-06-22:
This issue was moved to arduino/arduino-builder#163

closed 2016-06-22 by matthijskooijman