arduino/Arduino #5061: Reset while reconnect to WiFi

deadbringer17, 2016-06-21:
Hello everyone, 
I created a program that reads the pulses via Interrupt ( millis by storing in an array ) and send via wifi .
I have a problem , I do not understand why my program resets every time I turn off and turn on the modem... I use AP and  STA mode and i call this function in this order (in the example);

void setup(){
.......
WiFi.mode(WIFI_STA);
.....
WiFi.begin(ssid,pass);
.....
}

void reconnect(){
.......
WiFi.disconnect();
delay(100);
Wifi.begin(said,pass);
//wait wifi connect
......
}

void switchmodeAP(){
.........
WiFi.mode(WIFI_AP);
WiFi.softAP();
..........
.......
}

void switchmodeSTA(){
........
WiFi.softdisconnectAP();
WiFi.mode(WIFI_STA);
......
}

i call the WiFi.function in the right order?
reset can be caused by some error in the order of function calls?

thanks in advance

agdl, 2016-07-08:
Hi, which library and which shield are you using to do it?

sandeepmistry, 2016-07-18:
I'm closing this for now. Both the [WiFi101](https://github.com/arduino-libraries/WiFi101) and [WiFi](https://github.com/arduino-libraries/WiFi) libraries do not have a `WiFi.mode(mode)`, `WiFi.softAP()` or `WiFi.softdisconnectAP()` API's.

closed 2016-07-18 by sandeepmistry