arduino/Arduino #5063: Arduino 101 board

jimdgit, 2016-06-22:
How can I use c99 mode for this board?

facchinm, 2016-06-22:
Hi @jimdgit , which feature of C99 are you missing?
We are currently supporting C++11 which should contain C99 AFAIK
If you still need some C99 feature we are missing, please open an issue [here](https://github.com/01org/corelibs-arduino101) and you'll surely get a compete answer :smile:

closed 2016-06-22 by facchinm

jimdgit, 2016-06-22:
Ok sorry for posting to the wrong place.
I don't think any version of c++ can do what c99 does. For one thing, c89 does not allow variable declaration except at the beginning. I realize c++ would allow this but
I don't think any version of c++ allows Designated Initializers e.g.
struct point p = { .y = yvalue, .x = xvalue };
True?

PaulStoffregen, 2016-06-23:
You're going about this the wrong way.  Rather than ask these sorts of back-and-forth questions, you should post a complete program which can be copied and pasted into Arduino and causes a compile error when you click Verify.

matthijskooijman, 2016-06-23:
@jimdgit, if you need C99, try adding a .c file to your sketch. That will be compiled as C11, which should really contain all of C99 (unlike .ino files which are compiled as C++11).

jimdgit, 2016-06-23:
@matthijskooijman - I am using a .c file.
Here is what happens with a .c file:
void f()
{
  for( int i = 0 ; i < 10 ;++i )
  ;
}
Verify:
'for' loop initial declarations are only allowed in C99 mode

Oddly enough if I call it .cpp then not only does that compile but " Designated Initializers" seem to work as well.
What the mystery is now, is how come any of the 101 things work without any of the information in boards.txt or platform.txt in any file I can find. (from the corelibs-arduino101/boards.txt).
Thanks for the reply.

oqibidipo, 2016-06-23:
There is no `-std` option in `compiler.c.flags` in `platform.txt`. You can add `-std=gnu11` there, or better yet, copy that line to `platform.local.txt`and make your changes there.

matthijskooijman, 2016-06-23:
@jimdgit, what version of the Arduino IDE / core are you using? C11 got enabled in version 1.6.6 of the IDE for the avr core and in version 1.6.1 of the Arduino SAMD core (which I think the 101 uses? Or does it have its own core?).

As for the designated initializers, IIRC they are supported in C++ under some conditions (order must match declaration, they must be all be present, perhaps more), possibly as a gcc extension.

eriknyquist, 2016-06-23:
@matthijskooijman this is an x86 core (Arduino 101 board).

eriknyquist, 2016-06-23:
If anyone is still interested, the issue was resolved [here](https://github.com/01org/corelibs-arduino101/issues/220)