arduino/Arduino #5064: comment/uncomment keyboard doesn't work

cjdg, 2016-06-23:
Im using arduino on ubuntu studio 16.04 and there is an issue cannot use the keyboard shortcut to comment/uncomment

agdl, 2016-07-12:
@cjdg it works for me. Do you have this issue yet?

agdl, 2016-07-12:
The sortcut is CTRL + / BTW

NicoHood, 2016-07-12:
It does not work on german keyboard layouts as i remember, because there is no "slash" key. You have to press shift + 7 to get a slash. You can try that out yourself.

matthijskooijman, 2016-07-13:
On some non-standard layouts, this is indeed a problem. It is essentially a limitation in how keyboards are handled (in general and in java, I think). See https://github.com/bobbylight/RSyntaxTextArea/issues/157 for a more in-depth discussion of this problem.

per1234, 2021-03-29:
Thanks so much for reporting this bug @cjdg. Even though this was the original, there is slightly more information in the duplicate issue https://github.com/arduino/Arduino/issues/8090 that was created later.

It's best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.
If anyone has additional information to share, feel free to comment in the other thread.

closed 2021-03-29 by per1234

DenizMuhacir39, 2021-04-22:
i also cannot use ctrl+/. When i pressed them nothing happened