arduino/Arduino #5068: Add as snap package in Ubuntu

ericoporto, 2016-06-24:
It would be awesome if the Arduino IDE was available from Ubuntu Software Center. If it can't be available as a deb, maybe a snap package.

cousteaulecommandant, 2016-07-20:
It IS available on the Ubuntu repositories: http://packages.ubuntu.com/xenial/arduino -- however you probably don't want that version (1.0.5, wow, that's old...).  Debian Experimental has 1.5.6, which is not exactly bleeding edge either.  Apparently there are some license related issues slowing down the upgrade (see #4575).

I don't know how the Ubuntu software center works, but I would assume that if it's in the repositories it would show up in the software center.

mhall119, 2016-08-15:
I have a working snap package of 1.6.10, the snap packaging configs are here: https://github.com/ubuntu/snappy-playpen/tree/arduino/arduino

It builds from the release tarball, but it could just as easily use a local build directory (just change the source: for the 'upstream' part). I'm eager to help upstream adopt this, so if you're interested I will work with you to integrate it with your build infrastructure.

I have uploaded my snap package into the Ubuntu Software store under the name "arduino-mhall119" so that the "arduino" package namespace is still available to you. If you are on Ubuntu 16.04 or a distro with snapd installed, you can try it by running "snap install --devmode arduino-mhall119" (the --devmode is temporarly needed to access USB until the serial-port interface is fixed).

mhall119, 2016-08-15:
FWIW, if upstream produces snap packages, you can publish them directly into the store in a matter of seconds, you don't need to wait for it to go into Debian's archives, and then into Ubuntu's archive, and in then into a release. You can even push the nightly builds into the store (it has an "edge" channel specifically for this).

mlepage, 2016-12-17:
It would be awesome if we could get official stable and nightly Arduino IDE builds into Ubuntu Software Center, officially. What has to be done to make that happen?

NicoHood, 2016-12-18:
@miepage contact ubuntu or debian. Arduino is available for ArchLinux :P

Avamander, 2016-12-29:
This is an issue with licensing not being clarified. See https://github.com/arduino/Arduino/pull/2703

maqifrnswa, 2017-01-10:
I think the licensing issue is fixed, it can be reintroduced again. Snap packages are a good idea too if someone wants to do them.

pedzed, 2017-10-16:
@mhall119 Thanks for your snap, but could you update it? v1.6 is not supported by VSCode (which I am using to get rid of the default editor).

closed 2019-04-12 by ericoporto

Avamander, 2019-04-17:
So is this resolved?