arduino/Arduino #5071: Documentation Interrupts mainly DUE

techpaul, 2016-06-25:
# Interrupts on Due
Whilst the Arduion software for DUE supports interrupts on any pin, there is a hardware factor you have to take into consideration. Does that pin have a Schmitt trigger on its input?

So what you might say. Well if you are just polling digital inputs it does not matter, but when you are using interrupts this becomes important as the Schmitt trigger applies hysterisis to the input voltage, this means the input signal has to change by a larger amount before the change is seen by the hardware. This effectively gives faster edges on signal changes.

So if you have a slow edge on an input signal, especially from a switch it is very easy to get multiple interrupts from one slow edge. So using edge interrupts (RISING, FALLING, CHANGE), can be problematic, and in some cases this will also affect Level (LOW, HIGH) interrupts.

## Due Implications
On the Due, most of the input pins have Schmitt trigger however some are not !

The only place you can find the information currently is in the Atmel datasheet for the processor. It is there but you really have to positively look for it as in most places it is a general footnote, and only in detail really listed as footnotes to a table on Page 12 of the datasheet. The list of affected port pins is below

|Port Pin|Digital Pin|Named|
|--------|-----------|-----|
|**Port A**|||
|PA0|D69|CANTX0|
|PA9|D1|TX0|
|PA26|D75|MOSI|
|PA29|D87|PWM4|
|PA30|-|-|
|PA31|-|-|
|**Port B**|-|-|
|PB14|D53|CANTX1|
|PB22|-|-|
|**Port C**|-|-|
|PC2|D34|-|
|PC3|D35|-|
|PC4|D36|-|
|PC5|D37|-|
|PC6|D38|-|
|PC7|D39|-|
|PC8|D40|-|
|PC15|D48|-|
|PC16|D47|-|
|PC17|D46|-|
|PC18|D45|-|
|PC19|D44|-|
|PC20|-|-|
|PC21|D9|PWM9|
|PC22|D8|PWM8|
|PC23|D7|PWM7|
|PC24|D6|PWM6|
|**Port D**|-|-|
|PD10|D32|-|
|PD11|-|-|
|PD12|-|-|
|PD13|-|-|
|PD14|-|-|
|PD15|-|-|
|PD16|-|-|
|PD17|-|-|
|PD18|-|-|
|PD19|-|-|
|PD20|-|-|
|PD21|-|-|
|PD22|-|-|
|PD23|-|-|
|PD24|-|-|
|PD25|-|-|
|PD26|-|-|
|PD27|-|-|
|PD28|-|-|
|PD29|-|-|
|PD30|-|-|

Note that pin 4 PA29 (D87) is also shared with PC26 (D4) so using that pin as interrupt input will probably mean you can use edge interrupts on that actual pin.

I have a modified version of the unofficial DUE pinout diagram which I am contacting that maintainer for inclusion on their resource.
## Other Board Implications
As of yet I have not found details on what are schmitt or not input pins for Zero or 101, and have not looked at Uno and Mega processor datasheets
## Workarounds
1.  Use other pins if possible with schmitt trigger inputs for slow moving signals
2.  Put external schmitt trigger buffers or similar circuits externally.to provide fast edges
    *   Most standard resistor and/or capacitor debounce methods will not stop this problem. As they cause a slower edge and any noise will cause extra edge interrupts.

saponas, 2017-06-14:
I'd like to add that it would nice to see the interrupt documentation also state that you can't add multiple interrupt handlers on one pin on the DUE. A naïve reading of https://www.arduino.cc/en/Reference/AttachInterrupt might think you can add a "rising" ISR and a "falling" ISR on the same DUE pin.

techpaul, 2017-06-14:
That does not surprise me, it is the exception on most architectures to have more than one attached interrupt for different events on same pin or I/O event

Realistically I doubt the documentation will be updated for Due as it is now no longer available from normal sources.

per1234, 2019-02-05:
Moved to https://github.com/arduino/reference-en/issues/521

closed 2019-02-05 by per1234