arduino/Arduino #5077: build.path with relative path specified creates new sub sub sub directories on every run

plocher, 2016-06-27:
MacOS IDE environment
<code>$(ARDUINO) --pref build.path=./obj  --board $(BUILD) --verify CP_Sargent.ino</code>

after 4x invocations, the ./obj directory contains multiple nested instances of "obj/sketch"

<pre>
./obj/sketch
./obj/sketch/CP_Sargent.ino.cpp
./obj/sketch/CP_Sargent.ino.cpp.d
./obj/sketch/CP_Sargent.ino.cpp.o
./obj/sketch/obj
./obj/sketch/obj/preproc
./obj/sketch/obj/preproc/ctags_target.cpp
./obj/sketch/obj/sketch
./obj/sketch/obj/sketch/CP_Sargent.ino.cpp
./obj/sketch/obj/sketch/obj
./obj/sketch/obj/sketch/obj/preproc
./obj/sketch/obj/sketch/obj/preproc/ctags_target.cpp
./obj/sketch/obj/sketch/obj/sketch
./obj/sketch/obj/sketch/obj/sketch/CP_Sargent.ino.cpp
./obj/sketch/obj/sketch/obj/sketch/obj
./obj/sketch/obj/sketch/obj/sketch/obj/preproc
./obj/sketch/obj/sketch/obj/sketch/obj/preproc/ctags_target.cpp
./obj/sketch/obj/sketch/obj/sketch/obj/sketch
./obj/sketch/obj/sketch/obj/sketch/obj/sketch/CP_Sargent.ino.cpp
</pre>

Changing the CLI to use an absolute path is a workarround for this issue:
<code>$(ARDUINO) --pref build.path=$(PWD)/obj  --board $(BUILD) --verify CP_Sargent.ino</code>

matthijskooijman, 2016-06-27:
I believe the problem is not so much a relative path, but having a build path inside the source directory.

Also, this is a duplicate of https://github.com/arduino/arduino-builder/issues/89 which is already fixed in git master (and should be fixed in the next IDE release).

matthijskooijman, 2016-06-27:
Hm, wait, if you change it to an absolute path, it does work. Seems I read to quickly, this might be another issue after all.

I'm curious as to this is fixed with fixing the issue linked above.

@cmaglie, I think the commit to fix that arduino-builder issue is not yet present in an arduino-builder release, and thus also not in the Arduino IDE nightly build. Perhaps you could make an arduino-builder release so @plocher can check if this issue is fixed?

legionaryu, 2017-10-25:
I can confirm that the same thing is happening on windows when using the "arduino_debug.exe" from the command line on Windows.

It happens even with full path, from my own debugging, the issue is related to the folder content, e.g., if the build.path folder is empty, then it uses the folder as output for the binaries and etc, else it creates a new folder inside the folder "sketch" with the same name as the build.path folder, repeating these conditionals recursively.

My Arduino version is 1.8.3

Cheers

per1234, 2017-10-25:
@legionaryu it's not really helpful for you to do testing with outdated versions of the Arduino IDE. Currently, all testing should be done using the beta build: https://www.arduino.cc/en/Main/Software#beta.

legionaryu, 2017-10-25:
@per1234  I've just tested using the version 'arduino-PR-beta1.9-BUILD-22' and the problem still happens.
I'm executing the following command: arduino_debug.exe  --verify --board arduino:avr:uno c:\ArduinoTester\ArduinoTester.ino --verbose --pref build.path="c:\ArduinoTester\build"

You can see, in the image below, that 'arduino_debug.exe' keeps creating new 'build' folders recursively: 
[image]

legionaryu, 2017-10-26:
I was trying to find the issue in the source and I found the method [saveHex](https://github.com/arduino/Arduino/blob/7e2f7981f5fbb020b4de4057210ece113437ba77/arduino-core/src/cc/arduino/Compiler.java#L363), my 'spider sense' tells me that the issue has something to do with the conditional:
```
      if (Files.exists(compiledSketchPathInSubfolder)) {
        compiledSketchPath = compiledSketchPathInSubfolder;
      } else if (Files.exists(compiledSketchPathInBuildPath)) {
        compiledSketchPath = compiledSketchPathInBuildPath;
      } else {
        return;
      }
```

The line that creates new folders probably is this one [`Files.copy(compiledSketchPath, copyOfCompiledSketchFilePath, StandardCopyOption.REPLACE_EXISTING);`](https://github.com/arduino/Arduino/blob/7e2f7981f5fbb020b4de4057210ece113437ba77/arduino-core/src/cc/arduino/Compiler.java#L382)