arduino/Arduino #5079: Comments at end of concatenation macro cause error

jrt999, 2016-06-27:
A concatenation macro having a comment at end of same line causes Arduino IDE to fail - appears to be when it is "pasting" for indexing source code. CPP processes the file fine.

Example code

#define RXN 2  /* comment here causes error: pasting ... does not give valid preprocessing token */
#define RXN_BUFHEAD(X)        Rx ## X ## _BufHead
#define Rxn_BUFHEAD(X)        RXN_BUFHEAD(X)
#define Rxn_BufHead           Rxn_BUFHEAD(RXN)

static uint8_t   Rxn_BufHead;  // User read steps on, then takes from buffer

void setup() {

  Rx2_BufHead = 0;
}

void loop() {
}

thinkyhead, 2018-02-26:
Seeing this again in Arduino IDE 1.9 beta.
See the issue linked above, for example.
```
```
sketch\src\module\thermistor/../../inc/../pins/pins_RADDS.h:143:31: error:
pasting "/* BED*/" and "_WPORT" does not give a valid preprocessing token

 #define HEATER_BED_PIN      7 // BED
                               ^

sketch\src\module\thermistor/../../inc/../HAL/HAL_DUE/fastio_Due.h:75:33: note: in definition of macro '_WRITE'
   volatile Pio* port = (DIO ##  IO ## _WPORT); \
                                 ^

sketch\src\module\thermistor/../../inc/Conditionals_post.h:850:31: note: in expansion of macro 'WRITE'
   #define WRITE_HEATER_BED(v) WRITE(HEATER_BED_PIN, (v) ^ HEATER_BED_INVERTING)
                               ^

sketch\src\module\thermistor/../../inc/Conditionals_post.h:850:37: note: in expansion of macro 'HEATER_BED_PIN'
   #define WRITE_HEATER_BED(v) WRITE(HEATER_BED_PIN, (v) ^ HEATER_BED_INVERTING)
                                     ^

sketch\src\module\temperature.cpp:877:11: note: in expansion of macro 'WRITE_HEATER_BED'
           WRITE_HEATER_BED(LOW);
           ^
```

matthijskooijman, 2018-02-26:
Interesting isue. From the error in the last comment, it seems that the preprocessor is converting the `// BED` comment to `/* BED */` so it does not cause issues in most contexts, except for this one.

It does seem this is a compiler bug. From the C++11 (draft) spec (2.2.3):

> The source file is decomposed into preprocessing tokens (2.5) and sequences of white-space characters
> (including comments). A source file shall not end in a partial preprocessing token or in a partial com-
ment. Each comment is replaced by one space character.

This happens *before* processing preprocessing tokens, so the code shown should compile.

However, I'm not able to reproduce this error with any of the compilers I have at hand here. I do not have the 1.9 beta installed, but I do not believe the arm gcc version is changed there. @thinkyhead, could you show a more self-contained example (or test the example from @jrt999 in your setup) and provide some more details about your setup (what board/core are you using, what core version) and a full verbose compile output (enable in preferences) would also help in reproducing this.

thinkyhead, 2018-02-26:
Here's a branch of Marlin that should exhibit the issue on build (for RADDS),
assuming nothing else is exacerbating it:
https://github.com/thinkyhead/Marlin/tree/bf2_radds_pin_error