arduino/Arduino #5080: Please add ResponsiveAnalogRead library

dxinteractive, 2016-06-27:
Hi, I've made a library for filtering out removing noise from analog inputs. It's a little more sophisticated than a normal low pass filter / EMA and should be quite useful for applications where noise reduction and quick response times are both needed, or where the number of changes to the read value should be minimised over time. I have a blog post on how it works and perhaps a better reasoning [here](http://damienclarke.me/code/posts/writing-a-better-noise-reducing-analogread) if you're interested. Could I please get this added to the list of libraries? Please let me know if I need to modify anything in preparation or if you have any questions / concerns.

https://github.com/dxinteractive/ResponsiveAnalogRead

tofrnr, 2016-06-27:
on which algorithms is it based?
perhaps Median? or Kalman Filtering?

where is the documentation about the parameter "true"
ResponsiveAnalogRead analog(ANALOG_PIN, true);

- what is "false" instead" ?

agdl, 2016-06-27:
Hi, I'm sorry but your library can't be added since tag and library.properties are not aligned (version=0.0.1 differs from tag 1.0)

dxinteractive, 2016-06-27:
Hi @tofrnr - I didn't base this off an existing single algorithm, it was more an exercise in programming for a specific set of criteria (hence the detail in the blog post). It uses an EMA where the multiplier becomes much higher the larger the difference between the target and output values are, then an activity-timeout style sleep algorithm on top of that to optionally keep the number of output changes to a minimum. That's what the second constructor argument sets, whether to enable sleep or not.

Documentation about the constructor arguments is missing you're right. I'll add that back in. It does exist in the example *.ino and in the *.h file but should be in the README with everything else.

@agdl thanks for picking that up, I'll make that change soon.

dxinteractive, 2016-06-28:
Hi @agdl, I've amended the library.properties file and the README and bumped a patch version to 1.0.1.

agdl, 2016-06-28:
Done thank you!

closed 2016-06-28 by agdl

dxinteractive, 2016-06-28:
Thanks agdl!