arduino/Arduino #5084: config.h

xcvista, 2016-06-29:
I am here suggesting a possible way of allowing configuring the libraries in a reliable way.

Currently when a library requires configuration, it takes those information either as arguments in the initializer, or build it into the code as `#define`s. The first type is already customizable, but what about the latter, as used in, say, some variant of Ethernet library that can drive `W5100`, `W5200` and `W5500`?

I am suggesting the use of an optional `config.h` file. Only `#define`s are allowed in this file, and those lines are turned into `-D` flags that is present when compiling the entire sketch. This will allow some libraries to be configured at compile time.

per1234, 2016-06-29:
See https://github.com/arduino/arduino-builder/issues/15 and https://github.com/arduino/Arduino/pull/3757. I agree that something like this would be very useful but it has been requested multiple times in various forms(see also https://github.com/arduino/arduino-builder/pull/29, https://github.com/arduino/Arduino/pull/3717) and it seems this is not considered a desirable feature by the Arduino team(see https://github.com/arduino/arduino-builder/issues/15#issuecomment-145558252). I do understand the idea of forcing library authors to create a more user friendly API but sometimes you just need the preprocessor and I don't consider forcing the user to edit the library every time they want to change the configuration to be very user friendly.

I found it interested that .org did merge https://github.com/arduino/arduino-builder/pull/29.

xcvista, 2016-06-29:
@per1234 This is a bit disappointing to me somehow. Any good IDE should allow the users some way to manage this really.

I have a hobby project to rewrite the IDE from ground up (but optimize away a few problems like `keywords.txt` and a sketch preprocessing using a proper project file `project.ocxjson`) and in the project file I have an array called `/cpp/defines` that get converted into `-D`'s.

ameliaamelia, 2016-07-09:
All of us know that in a library module we can insert i.e. the following include:
`#include "X:\Project\Config.h"`
that works but at this point the library module **is no more a library module**.
But if there was a Common Macro like **$(TargetDir)** or **TARGETDIR** to write:
`#include "X:\$(TargetDir)\Config.h"` or `#include "X:\TARGETDIR \Config.h`"
everything would be fine, side effects; at worst if the Config.h is not present in the project directory, we would have an error message.

I recently downloaded a library very well done with a huge amount of features.
I customized a config.h for each project manually (silly) insert in the library directory.
I got a reduction of 2k-4k depending on the unused functions.
By working with the micro every byte saved is a good thing.

amelia

facchinm, 2019-09-19:
Closing this as the discussion about the metadata (or configuration) file is going to take place in [arduino-cli](https://github.com/arduino/arduino-cli) repo and in the mailing list

closed 2019-09-19 by facchinm