arduino/Arduino #5085: issue with atmega328p xplained mini and arduino 1.6.9

dave2x11, 2016-06-29:
I can program the mini board using Atmel Studio 7.0 and the USB micro connector and everything works fine.

When I try to use the Arduino 1.6.9 IDE , I can select the board and the correct com port but the program won't load and I get the error:      avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0xe0 etc.

carlosperate, 2016-06-29:
You need to burn the bootloader first: https://github.com/AtmelUniversityFrance/atmel-avr-xmini-boardmanagermodule/wiki/01.-Preparing-atmega*-Xplained-mini-for-Arduino-IDE-use-%28MS-Windows%29

dave2x11, 2016-06-29:
If I burn this bootloader then the arduino IDE will work right?

If I do this will Atmel Studio still work through the micro USB port?

Or is it I can have one or the other but not both?

Of course, Atmel studio can always be used with a programmer through the 6 pin ISP connector. But then I also lose the bootloader.

David

facchinm, 2016-07-19:
Hi David,
you can keep compatibility with both AS and ArduinoIDE by taking care not to erase all chip when loading from AS and load your binary into application space (also the fuses should be set accordingly).
@aethaniel, am I missing anything?

aethaniel, 2016-07-19:
@facchinm, it should work indeed like that. @dave2x11, can you confirm AS doesn't erase the bootloader when flashing the application binary, please?
if the bootloader is optiboot, you can check the fuses parameters here: https://github.com/AtmelUniversityFrance/atmel-avr-xmini-boardmanagermodule/wiki/01.-Preparing-atmega*-Xplained-mini-for-Arduino-IDE-use-(MS-Windows)

carlosperate, 2016-07-19:
Sorry for the diverting question, but does that mean that the Arduino bootloader will only flash and erase the application flash section? So it doesn't matter how big/small the bootloader is, none of the bootloader flash section is used for the sketches?

aethaniel, 2016-07-19:
@carlosperate, AFAIK, if the fuses configuration matches the bootloader size you will be safe.

carlosperate, 2016-07-19:
Thanks for the response @aethaniel. I was mostly wondering if the bootloader size had any implications at all on the space available from sketches, since I was under the impression that ages ago some alternative bootloaders used to claim that their smaller size allowed for more space for the sketch. But this is an area I don't really know much about, so I guess that would only be relevant for microcontrollers without a boot flash section (attiny/atmega48 most likely?).

aethaniel, 2016-07-19:
@carlosperate, I'm not at all AVR expert and use since some time Atmel AVR atmega328p Xplained minis with optiboot bootloader which weights 512 bytes. So, I don't have know anything about the AVR bootloader history. I'm more experimented on ARM Cortex-M devices :yum: 
@facchinm could answer certainly much better than me on this subject.

facchinm, 2016-08-26:
According to http://www.martyncurrey.com/arduino-atmega-328p-fuse-settings/ , older bootloader size was 2K, while 0.5K is the minimum. Setting BOOTSZ fuses you can tune the size and get additional space for the sketch (this also applies to 328p).
[image]

Closing the issue as solved

closed 2016-08-26 by facchinm