arduino/Arduino #5086: Minor Typo in sketch naming error message.

ZoogerZager, 2016-06-29:
[image]

The message should read:

The sketch name had to be modified. Sketch names can only consist
of ASCII characters and numbers (but cannot start with a number).
They should also be less than 64 characters long.

per1234, 2016-06-29:
You should check the hourly build to make sure the issue hasn't already been fixed. I don't see that typo on the hourly build or 1.6.9.

sandeepmistry, 2016-06-29:
@per1234, thanks for checking this in the hourly build!

@ZoogerZager based on your screenshot, you using version 1.0.5 of the IDE which is really old. I would suggest you upgrade to 1.6.9.

closed 2016-06-29 by sandeepmistry