arduino/Arduino #5089: "bossac: extra arguments found" when uploading to Due with 1.6.8 board

archenemies, 2016-07-01:
Hi,

I'm somewhat new to hacking on Arduinos. I got a RedBoard and it worked fine, but the first big hurdle I encountered when experimenting with my new Due was that I couldn't upload sketches with the latest software.

In the Arduino IDE with SAM board 1.6.8 selected in the Boards Manager, I get this error

    Sketch uses 22,540 bytes (4%) of program storage space. Maximum is 524,288 bytes.
    bossac: extra arguments found
    Try 'bossac -h' or 'bossac --help' for more information

when trying to upload e.g. the ASCIITable sketch. I had to downgrade to the 1.6.7 board to get it to work.

After a few months I noticed that I still get the same error. Could you either fix the problem or let people know that there is one? I thought Arduinos were supposed to be easy for beginners...

https://www.arduino.cc/en/Guide/ArduinoDue

I'm using Arch Linux, I have installed the arduino package from AUR.

facchinm, 2016-07-01:
Hi fellow Archer :smile: 
Could you post the full verbose output of the failing upload?

archenemies, 2016-07-01:
Wow, I've never bonded with someone over using Arch before.

I guess I shouldn't have assumed you already knew about the problem.

I'm attaching the output of:

    arduino --upload scopeserver/scopeserver.ino --port $(get-prog-port) --verbose |& tee upload-verbose.txt

[upload-verbose.txt](https://github.com/arduino/Arduino/files/344284/upload-verbose.txt)

TIA

facchinm, 2016-07-04:
In fact, you are getting a variable replaced in the wrong way
```
bossac -i -d --port=ttyACM5 -U false -e -w true -b ....ino.bin -R 
```
should be 
```
bossac -i -d --port=ttyACM5 -U false -e -w -v -b ....ino.bin -R 
```
With my configuration (IDE for git, SAM core v 1.6.8) it works on both Programming and Native port.
Which version of the IDE are you using?

archenemies, 2016-07-04:
$ arduino --version
    Picked up JAVA_TOOL_OPTIONS:
    [3]$ man arduino
    No manual entry for arduino
    [16]$ arduino --help
    Picked up JAVA_TOOL_OPTIONS:
    [3]$

I think it is a recent version, I downloaded it about three months ago? The AUR package directory has a file named `arduino-1:1.6.8-2-x86_64.pkg.tar.xz` and one named `arduino-1.6.8-linux64.tar.xz`.

facchinm, 2016-07-04:
Would you mind testing with the official version from [https://www.arduino.cc/en/Main/Software#hourly](https://www.arduino.cc/en/Main/Software#hourly) ?

archenemies, 2016-07-04:
It seems to work with both the official nightly version, and the version in AUR. Thank you!

closed 2016-07-05 by facchinm