arduino/Arduino #5092: Windows: incompatible JNA native library installed

jczimm, 2016-07-03:
**Windows 10
Arduino 1.6.9**

---

`arduino.exe` closes shortly after execution.

Running `arduino_debug.exe`:

```
arduino-1.6.9$ arduino_debug
java.lang.Error:

There is an incompatible JNA native library installed on this system
[the beginning of my PATH]
To resolve this issue you may do one of the following:
 - remove or uninstall the offending library
 - set the system property jna.nosys=true
 - set jna.boot.library.path to include the path to the version of the
   jnidispatch library included with the JNA jar file you are using

        at com.sun.jna.Native.<clinit>(Native.java:146)
        at com.sun.jna.Pointer.<clinit>(Pointer.java:41)
        at com.sun.jna.platform.win32.WinReg$HKEY.<init>(WinReg.java:32)
        at com.sun.jna.platform.win32.WinReg.<clinit>(WinReg.java:61)
        at cc.arduino.os.windows.FolderFinderInWindowsRegistry.findInternal(FolderFinderInWindowsRegistry.java:53)
        at cc.arduino.os.windows.FolderFinder.find(FolderFinder.java:48)
        at processing.app.windows.Platform.recoverSettingsFolderPath(Platform.java:56)
        at processing.app.windows.Platform.init(Platform.java:48)
        at processing.app.Base.guardedMain(Base.java:153)
        at processing.app.Base.main(Base.java:135)
```

I took a look at #1948 and #2301 but could not figure out how to set the system property described.

facchinm, 2016-07-04:
Hi @jczimm , the problem is indeed #1948, which was fixed by #2301 but reintroduced by the twin commits https://github.com/arduino/Arduino/commit/f6546aea4527bb0723fac6472c1e10328ebe9473 and https://github.com/arduino/Arduino/commit/98eb1a9ee22f3b346787ff9608ab8fa7a7271e6e

@cmaglie , can we safely reapply https://github.com/arduino/Arduino/commit/e7fc9a1bd24bb6d666f3ad9d102607941fa0d29c ?

facchinm, 2016-07-04:
@jczimm , I just opened a PR here https://github.com/arduino/Arduino/pull/5097.
ArduinoBot should build it and place the compiled packages as a comment. Could you test the Windows one and report if it solved your problem? Thanks

jczimm, 2016-07-04:
@facchinm , this solved the problem! Thanks!

PR #5097 (2016-07-05): Windows: addind -Djna.nosys=true to launcher options
reapply patch #2301
this patch got lost with f6546ae and its partial revert 98eb1a9

Should fix #5092
commit 76029ae5 (2016-07-05): Windows: addind -Djna.nosys=true to launcher options

reapply patch #2301
this patch got lost with f6546ae and its partial revert 98eb1a9

Should fix #5092
closed 2016-07-04 by jczimm