arduino/Arduino #5093: Wrong symlink causes compiling failure

NicoHood, 2016-07-03:
```
stat /home/arch/hackallthethings/sketchbook/hardware/arduino3/avr: no such file or directory
```

The path above does no longer exist. avr points to an inaccessible folder (cause I swtiched my OS). But this error caused the whole ide to not work. This is not a normal case, but should not cause the whole IDE to not function.

matthijskooijman, 2016-07-03:
"caused the whole ide to not work" <-- that's a bit of a lousy error description, you really should know better than that. Does it not start? Does it print anything on stderr? Is there an error popup? ALso, is this `sketchbook` folder configured as your sketchbook?

NicoHood, 2016-07-04:
Yes it is configured as sketchbook and sorry, the uploading did not work anymore.

facchinm, 2016-07-19:
Should be fixed by https://github.com/arduino/arduino-builder/pull/166

NicoHood, 2016-09-17:
I have not tested this again but I'll close it of you say so.

closed 2016-09-17 by NicoHood