arduino/Arduino #5094: libncurses error for bundled avrdude on Arch

NicoHood, 2016-07-03:
On arch I get:

```
/usr/share/arduino/hardware/tools/avr/bin/avrdude: error while loading shared libraries: libncurses.so.5: cannot open shared object file: No such file or directory
```

I used my own PKGBUILD to build arduino:
https://github.com/arduino/Arduino/issues/5082

I wanted to use the built in avrdude, as several libraries/cores rely on this specific version and not arch linux upstream. However it seems that avrdude does not run like this on arch. **This has nothing to do with my PKGBUILD, as this also happens with the normal zip file download.** This should be fixed ASAP, as its incompatible with arch.

It would also be nice if you can comment on the linked issue. I want to know if using upstream deps (avrdude, java) will probably cause problems. If not I could go the way the official AUR package does.

matthijskooijman, 2016-07-03:
The obvious question is: Do you have (the right version of) libncurses installed? If not, I guess that's the thing to fix (or the zipfile should ship a libncurses version, but I'm not sure if that would tie in neatly with the system's libraries then).

As for using upstream tools, the Debian package uses upstream avrdude and avr-gcc, though that required some changes in Arduino (and they only have 1.0.x so far). For 1.5+, with platform.txt, using the system tools is more complicated, especially when third party cores are involved, or when cores are updated by the Arduino IDE.

facchinm, 2016-07-04:
The avrdude currently packaged relies on `libncurses5`, while new avrdude https://github.com/arduino/avrdude-build-script is statically compiled without that dependency.
The `libncurses` devs changed the name of the library while updating to version 6, thus you need to add libncurses5 (and probably libtinfo5) as dependencies in your recipe.
Closing as duplicate of https://github.com/arduino/toolchain-avr/issues/25

closed 2016-07-04 by facchinm

NicoHood, 2016-07-04:
But `libncurses5` is not available on arch.

Only this is available but does not work:
```
core/ncurses 6.0-4 [installed]
    System V Release 4.0 curses emulation library
```

facchinm, 2016-07-04:
You can get it from AUR (I also believe that `libtinfo-compat` existed at some point in the official repos but I can't find it anymore)

NicoHood, 2016-08-22:
Please reopen, as the revert of avrdude 6.3 made the issue valid again.