arduino/Arduino #5096: Arduino IDE under Linux serials port list

stowite, 2016-07-04:
Since about 1.5.0 under Linux the drop down list of serial ports has been filtered in such a manner that one cannot use device names created using udev. At one time one could create a port using udev definitions such as 
KERNELS=="1-1.1.3:1.0", SYMLINK+="ttyAtmega328P", GROUP="dialout"

For a later release I had to change to use udev definitions such as
KERNELS=="1-3.4.4:1.0", SYMLINK+="ttyUSB99-stepper", GROUP="di
alout"
the important point being that the device name had to be prefixed by a something that could be matched by the Java regular expression hard coded into the IDE source code. The latest release has gone one step further in the restrictions so that I can no longer use udev to define a device name.

I decided to try to find the source of the problem so built the IDE from a GIT checkout. No problem with this so I then searched though the code to find the regular expression involved. It seemed to be in Java class processing.app.SerialPortList so I modified it and re-built.This built OK but nothing changed! Debugging I found that this class is not used to populate the port list. I then went to the JSSC source code and found the identical regular expression as in class  processing.app.SerialPortList!  I modified this and re-built. Again this built OK but nothing changed. Doubting my sanity I debugged again and found the the list is populated by a native method listSerialsNative() which does not seem to be part of JSSC. I cannot find the source code for this but it does seem to have the same regular expression as both JSSC and class processing.app.SerialPortList.

Somewhere along the line the IDE stopped using RXTX and went over to using JSSC ( a move I approve of since it works more reliably than RXTX ) BUT it seems that at that point the strict serial port name restrictions came in.

Why three identical regular expressions for the filter with two of them not being used and the one that is being used in a native method that does not seem to have it's source in GIT. To my mind a filter regular expression should be used but this should be in the configuration file and not hard coded.

If this issue is not currently being addressed I shall spend more time on trying to fix this but it could be some while before I get round to it.

facchinm, 2016-07-04:
Hi @stowite , IDEs from 1.6.8 use a different approach for serial ports listing. In fact, an external library (https://github.com/arduino/listSerialPortsC) is in charge of listing the serial ports, associating with the correct VID/PID to get the port names in the IDE's menu.
The entry point is [here](https://github.com/arduino/Arduino/blob/master/arduino-core/src/cc/arduino/packages/discoverers/serial/SerialBoardsLister.java#L58), and the native method is called [here](https://github.com/arduino/Arduino/blob/60b169fbe93c861b6c06d57f57b8f5b2be5919e0/arduino-core/src/processing/app/Platform.java#L185)

I believe that symlinking the port should be ok with this method but I'm not sure about that.
Hope these info are enough to let you start hacking :wink:

stowite, 2016-07-05:
Thanks for the response. I have followed the delegation chain down - 
cc.arduino.packages.discoverers.serial.SerialBoardsLister delegates to JNI library listSerialPortsC which delegates to the cross platform C library libserialport which delegates to the list_ports() method in linux.c. The list_ports() method iterates through soft links in directory "/sys/class/tty"  extracting tty devices. Unfortunately the soft links generated by udev do not appear in  "/sys/class/tty" ! This means that the current method of delegating serial port enumeration through to libserialport can never pick up the symbolic links generated by udev! Interestingly the FreeBSD version does seem to enumerate the ports defined in "/dev" !

In the short term I have ameliorated my udev problem by modifying Platform.listSerials() so that it delegates the enumeration back to jssc.SerialPortList.getPortNames() . The only way I can see the udev problem every being fixed is if one has a configuration parameter to indicate which delegation to use but I suspect this approach would not be popular!

stowite, 2016-07-06:
Having thought about this problem I believe that the failure to list the soft links created by udev is not just a limitation but is actually a bug. The bug was not there prior to the change to use libserialport. How do I get this raised as a bug?

facchinm, 2016-07-13:
You could open a bug in https://github.com/arduino/listSerialPortsC, attaching the output of `listSerialC` executable (get it from [here](http://downloads.arduino.cc/liblistSerials/liblistSerials-1.1.2.zip)) launched with `LIBSERIALPORT_DEBUG=1` environment variable set

jkorinth, 2017-12-31:
Any news on this one? This thread saved me at least from recompiling the IDE, but the problem persists for me. Tried creating a manual symlink in `/sys/class/tty`, but that does not seem to be possible. Considered writing a kernel module for a sec, but there has to be some sort of sane solution?

OH1KH, 2018-08-07:
I'm also looking for solution for this. Very annoying go through all serial HW devices after PC start to find Arduino. It was lot better when symlink was working.
IDE 1.8.5  in Fedora 27
Should I downgrade so many versions that I get back to RXTX ?
This seems to be open issue for years!

If symlink can not be used, develop a libserialport version that polls every physical port and finds where arduino/esp is and selects that at IDE start without user action.

Thank you!

anymouse404, 2020-09-25:
My issue #10782 was that I was not able to select a virtual serial ports (which were named /dev/tnt*). I find it bad practice to be going around /dev/ creating symlinks.
It would be much more convenient to just allow entering any file path. This is how almost any other linux software deals with serial ports.

mechsin, 2021-02-12:
So with some help from the folks on the dev mailing list I traced this bug down to how libserialport which is part of a larger project call Sigrok. I was able to make changes to libserilport that allows symlinks to be passed though and I have created a pull request with the libserialport project to get those changes merged. I was then able to get a version of listSerialPortsC to build with my changes to libserialport which in turn Arudino IDE then uses listSerialPortC to list the ports in the menu. Then building the IDE with my changes I was able to get my UNO board to show up in the menu at /dev/arduino and I was able to program it as well. I am confident that the changes I made to libserialport shouldn't effect functionality on Mac or Windows. I am less sure about BSD. There could be edge case issues with the changes for Linux but some more testing probably needs to be done.

The changes I made to the libserialport library can be found in my fork of the sigrok project at the following URL: https://github.com/mechsin/libserialport-1/tree/symlink_b

maxpolzin, 2022-08-17:
Dear mechsin,
Thanks for figuring all this out. The approach you describe sounds very intersting to me. 
Could you provide some more details how you managed to compile listSerialPortC and link against your changes in libserialport?

mechsin, 2022-08-19:
Max

Sorry for the delayed response I did that work a while ago and don't
remember all the details but let me take a look when I have time this
weekend and I can give you some more information

maxpolzin, 2022-08-19:
After I warmed up the thread after more than a year, I don't consider 1 day a delayed response at all. Thanks for your response.

Any additional info is appreciated, but please don't feel obliged to go through the whole process again. I can also figure out the details on my own.

mechsin, 2022-08-20:
Max,

It looks like it was actually relatively simple to add the libserial
modifications in. When I did this I was working on code from 1.8.13 but
there is really only one change you should need to make so it should matter.

First you need to modify the build.xml in the build folder. You need to
find the line that references the libserialport version. For me this was
around line 104. Update the text here to whatever version you decide to
make your new build with. It doesn't have to be a number, for instance I
call my version symlink.

After that you just need to place your version of the libserial library in
the build folder. The build tools are expecting the library to be in a zip
file. I just copied the folder structure of the original libserial library.
You also have to produce a SHA hash file for your zip. After that you can
just follow the instructions
<https://github.com/arduino/Arduino/wiki/Building-Arduino> on the Arduino
web site to build the Arduino IDE. If you did everything correctly then the
build process will pick up the new library and the sylinks should work in
your new build.

I attached my files if your using a 64 bit Linux system they should work
for you or you can use them as a example.

Chris

maxpolzin, 2022-08-21:
Amazing, thx for the detailed instructions, Chris. 
With your clarifications, I will be able to build the Arduino IDE with your modifications.

One more question:
"I attached my files if your using a 64 bit Linux system they should work for you or you can use them as a example."
I don't see any attached files. Did github maybe filter them?

mechsin, 2022-10-11:
Yeah the mail server probably stripped them off I forgot that mailing lists
like this usually don't allow attachments.

Let me know if you have issues and I can see what I can do.

Important note this will only work with the 1.x Arduino IDE. The new 2.x
IDE uses the as arduino-cli to do most of the work and that is written in
Go and doesn't use the library in question. There was a request that I add
the same functionality to it but I don't know Go and didn't have time. But
recently I have been thinking about getting back around to it.

Chris